OnionUI/Onion #1229: Issue Title

archcst, 2023-10-22:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

v4.2.1

### Miyoo Firmware Version

202306111426

### Provide a clear and concise description of the issue

Simplified Chinese rom name can't show correctly in Game Switcher.

### Steps to Reproduce

1. refresh roms.
2. press menu button to show Game Switcher.
3. Simplified Chinese rom name partly missing.

### Provide any additional context or information that may help us investigate your issue

[image]

fengfrw, 2023-10-22:
Modify the theme font or use a theme that supports Chinese perfection.（修改主题字体或使用支持中文完善的主题）

yyoshisaur, 2023-10-22:
It's not just in Chinese; Japanese titles are also missing.
This issue occurs even with fonts that support Japanese.

I'm attaching the "cache6.db" file.
The "disp" column appears to be correct, but it seems like the strings in the "pinyin" column (and "cpinyin" column) are missing.
I believe that the game titles displayed in GameSwitcher may be referencing the "pinyin" column.

I hope this comment helps.
[MD_cache6.zip](https://github.com/OnionUI/Onion/files/13062926/MD_cache6.zip)

aleksei74, 2023-10-28:
Korean is same.
It occurs in activity tracker too.
Korean and Japanese character save to space(0x20) in cache6.db file (pinyin and cpinyin)

maxpereira, 2024-01-18:
This issue seems to be more generalized, it appears to be most non-ASCII characters that are not rendering correctly, and I observe this behavior in several different apps. For example the game "Pokémon Sapphire Version" shows as "Pok mon Sapphire Version" in both GameSwitcher and ActivityTracker.

As @yyoshisaur mentioned, this is presumably because these apps are pulling from the "pinyin" or "cpinyin" columns in the cache DB however non-ASCII characters seem to be stripped (they show correctly in the "disp" column). I tried fixing the "pinyin" and "cpinyin" columns manually but the changes are reverted when the system starts up. Furthermore I am not sure this issue is even resolvable from within Onion as I think this might be handled by the underlying firmware (unless the individual apps can somehow be modified to read the name differently).