OnionUI/Onion #1247: Arcade MAME 2003 Plus not working

Goodingn, 2023-10-27:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2.2

### Miyoo Firmware Version

202204190828

### Provide a clear and concise description of the issue

Starting any arcade rom shows blank black screen and no sound.

Starting a game previously started on 4.1.2 there is sound, but screen is blank.

Deleting the 'auto' save state then trying gets blank screen and no sound.

Loading a saved state screen remains blank (doesn't display loaded state text), but sound happens. When going to Retroarch menu the loaded state text then also shows.

Restarting from Retroarch menu doesn't fix it and the Retroarch menu stays on the screen.

Exiting Retroarch from menu generates the error 'program exited unexpecte...' and '139'. The rest of the error is off the screen.

### Steps to Reproduce

As above.

### Provide any additional context or information that may help us investigate your issue

.

Phanturian, 2023-10-27:
Save states do not work properly with MAME 2003 Plus, or most Arcade games. Launch a MAME 2003 Plus game, and go into RetroArch Main Menu>Settings>Saving and disable Auto Save State and Load State Automatically, then save content directory or core override. FBNeo and FBA 2012 have better success with save states, but there are still issues. If there's an Arcade game you really need to use save states with, try using a FBNeo or FBA 2012 version instead.

Goodingn, 2023-10-27:
Thanks, but it's not the save states I have an issue with- though they worked fine on 4.1.3.

The games just don't start. And as Mame2003plus is the default, this is an error in the shipping of 4.2.2.

laesetuc, 2023-10-31:
Have you tried resetting settings?  Worth a try...

Goodingn, 2023-11-02:
> Have you tried resetting settings? Worth a try...

Yeah, that sorted it thanks.

Goodingn, 2023-11-09:
Found out that I can reproduce this issue by setting frameskip to any of the 'auto' values.  A frameskip of 1 works fine, any of the 'auto' frameskips shows no display.

schmurtzm, 2023-11-10:
> Found out that I can reproduce this issue by setting frameskip to any of the 'auto' values. A frameskip of 1 works fine, any of the 'auto' frameskips shows no display.

This is a known bug : switching audio to SDL should fix the problem with frameskip auto on Mame 2003 plus. This bug seems related to oss + frameskip (same problem on pcsx rearmed core if OSS audio output is selected).
Check in `Settings -> Audio -> Output` if `SDL` is selected and not in `OSS`.

Normally it's the [default sound option in Onion](https://github.com/OnionUI/Onion/blob/main/static/configs/Saves/CurrentProfile/config/MAME%202003-Plus/MAME%202003-Plus.cfg#L1).