OnionUI/Onion #1260: Netplay and Scrapper appears twice

kjm1988, 2023-11-03:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2.3

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

Netplay options appear twice upon pressing Y on any game, so does the Scrapper tool which appears both about halfway down and at the very bottom.

Netplay options under "Netplay (core name)" function as expected but the "Easy Netplay host, join and Pokémon" options in to root of the menu need removing as they take you back to the game selection screen.

### Steps to Reproduce

Press Y on any game, the nested Netplay options function as expected but the ones on the root of the menu don't.

### Provide any additional context or information that may help us investigate your issue

.

schmurtzm, 2023-11-10:
There is normally [a cleaner script](https://github.com/OnionUI/Onion/blob/a9037051623d705dbcfe5955df5440c72948dc98/static/build/.tmp_update/script/migration/00016_remove_old_glo_entries.sh) during the update to avoid these doublon. I don't know why it hasn't been executed on your device.

So you can remove these files :
```
SDCARD/App/romscripts/emu/Netplay.sh
SDCARD/App/romscripts/emu/Scraper.sh
SDCARD/App/romscripts/rapp/Scraper.sh
```

kjm1988, 2023-11-10:
@schmurtzm Thanks I will remove those manually. For info, this occured on two MM+'s both doing OTA update from 4.2.2 to 4.2.3 so I imagine this has happened to everyone else doing the same.