OnionUI/Onion #1261: Canceling search - Program exited unexpectedly

sanderDijkxhoorn, 2023-11-03:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2.3

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

Exiting the search form results in a unexpected error

### Steps to Reproduce

Press X (top button) to search and when the search term input comes up just press B (bottom button) to cancel and you will be seeing the screen “The program exited unexpectedly. (Error code: 139)”

### Provide any additional context or information that may help us investigate your issue

I update using OTA updater to 4.2.3

sanderDijkxhoorn, 2023-11-03:
[image]

sanderDijkxhoorn, 2023-11-05:
There is also a kind of similar issue reported on the MiyooMini Reddit: [Reddit post](https://www.reddit.com/r/MiyooMini/s/EjR7Pvlxl3)

sanderDijkxhoorn, 2023-11-05:
I switched my theme from this Cyber one back to the onion default and then again back to cyber one and it works fine again.

closed 2023-11-05 by sanderDijkxhoorn

Aemiii91, 2023-11-05:
So weird.. glad it's working now though