OnionUI/Onion #1262: Guest mode not working in 4.2.3 MM+

rohitpaul, 2023-11-04:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2.3

### Miyoo Firmware Version

202303262339

### Provide a clear and concise description of the issue

When you click on guest mode to turn it on or off, it doesn’t work. The screen changes to loading and then back to the menu, but the profile remains the same. 

### Steps to Reproduce

Guest Mode [ON]

Click

Guest Mode [ON]

### Provide any additional context or information that may help us investigate your issue

Just installed update.

schmurtzm, 2023-11-10:
Hi, I can't reproduce one my 4.2.3. Looks like SD card corruption imo.
Can you check that your file SDCARD/App/Guest_Mode/launch.sh  looks [like this](https://github.com/OnionUI/Onion/blob/main/static/packages/App/Guest%20Mode/App/Guest_Mode/launch.sh) ?
If not then uninstall and reinstall "Guest Mode" app from "Package Manager" (or do that directly and then test again).

rohitpaul, 2023-11-11:
Might be corruption per the comment above.

Completed the following steps:
  Put SD card on to Mac and ran disk utility repair - Issue persisted. 
  Checked launch.sh - looked to be the same.
  Previously uninstalled and reinstalled via package manager - issue persisted
  Reinstall Onion OS (download via mac and install by coping to SD card and letting it reinstall.) - issue persisted

rohitpaul, 2023-11-11:
Reformatted SD card and reinstalled the Os. Working now.

closed 2023-11-11 by rohitpaul