OnionUI/Onion #1263: Wifi-Teweaks not working

Laserstrahl, 2023-11-04:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2.3

### Miyoo Firmware Version

20230628128

### Provide a clear and concise description of the issue

Anychanges on the Tweaker->Network Menu are lost when closing the Tweaker app. Also the Services like http/smb/ftp are not started.

### Steps to Reproduce

Connect the Miyoo Mini Plus to the wi-fi network via:
Settings --> Wifi

Go to:
Apps › Tweaks › Network › 
Enable SMB or FTP or HTTPS... close the app.
non of the protocols are working for remote access to the Miyoo.

Go to:
Apps › Tweaks › Network › 
And see that the enabled settings are disabled again.

### Provide any additional context or information that may help us investigate your issue

Clean installation of release 4.2.3 on a new device with firmware v 20230628128
no roms added yet, nor any files modified.

vonblubba, 2023-11-08:
Same issue here.

XK9274, 2023-11-08:
You both got the file:

/mnt/SDCARD/.tmp_update/script/network/update_networking.sh ?

What are the EOL on the file?

By clean install did you do a drag and drop on a new SD?

vonblubba, 2023-11-08:
File `update_networking.sh` is present. LF EOL.
In my case, I performed an OTA update.

Laserstrahl, 2023-11-11:
File also present, also LF EOL.
I performed a new installation on a new SD Card.

XK9274, 2023-11-17:
I've done 2-3 installs on a fresh format of an SD card and I'm yet to have this problem. 

Can you run the snapshot function in Tweaks -> Advanced -> Diagnostics -> Snapshot

You'll get a file called log_export.7z in your SD, please send me that by DM on discord

My user is:

`_xk_` 

[image]

vitorabf, 2023-12-29:
Have you guys managed to solve the issue? I'm facing the same problem.

vonblubba, 2023-12-30:
It started working again for me. I did't touch anything so I do not have a solution to offer, sorry.
One thing I can suggest: be sure to completely exit the configuration menu. Wifi gets enabled only after you exit config.

XK9274, 2024-01-13:
For anyone still having this issue, please try beta 4.3

If you have the issue after that, Tweaks -> Advanced -> Reset settings and reset MainUI. If it persists after, go to Tweaks -> Advanced -> Diagnostics and generate a log snapshot and send me it.

closed 2024-06-23 by Aemiii91