OnionUI/Onion #1264: Latest Miyoo Mini V4 firmware (1027) causing stuttering issues on every emulator I've tested within OnionOS

Fendera87, 2023-11-04:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2.2 or 4.2.3 (Stable)

### Miyoo Firmware Version

202310271401

### Provide a clear and concise description of the issue

Hello everyone

### Description of the issue:

A couple of days ago, I applied the latest **Miyoo Mini V4 firmware 1027** that came out recently. After that update, every emulator I've tested on OnionOS introduced frame pacing issues, causing noticable stuttering.

After noticing this, I did a fresh OnionOS install on a wiped SDCard, using stock RetroArch settings but the stuttering issues didn't go away. 

### Examples:

**PS1:** 

- In the Tekken 3 character select screen menu, you can clearly see that the background (upwards stream) has some frame pacing issues. It's also noticable during gameplay. This wasn't the case before the update Miyoo Mini firmware update. 

**SNES:**

- In Super Mario World, the stuttering became very distracting during gameplay. 

### Steps to Reproduce

Updating a Miyoo Mini V4 to the firmware 1027 and using a freshly installed version of OnionOS 4.2.2 or 4.2.3. A good game to test is the PS1 game Tekken 3, there you can see in the character select menu, that the background upwards stream has some frame pacing issues. It should however be a fluid 60 fps motion without any stuttering. During gaeplay the same issue.

### Provide any additional context or information that may help us investigate your issue

I've also tried different SD Cards, nothing changed.

tGecko, 2023-11-04:
1027 firmware is not yet supported by Onion. I suggest you downgrade to a supported firmware.

Fendera87, 2023-11-04:
Do you know where I can find the 0611 firmware? There's only the 0612 version available on the Miyoo Mini site.

tGecko, 2023-11-04:
That's the correct one, it's just wrongly labeled on their website

Fendera87, 2023-11-04:
I did a downgrade to the 0611 firmware. The frame pacing issue is still there. Fresh OnionOS install but nothing helps at all. 

Only thing that came to my mind is enabling V-Sync in the RetroArch settings... I did this before downgrading the firmware and it had no effect for whatever reason (maybe I'm wrong). Now after the firmware downgrade, enabling V-Sync got rid of the frame pacing stuttering. 

I remember when first using OnionOS after I received my unit, V-Sync was disabled by default and there wasn't any stuttering going on at all. 

I'm so confused.

Fendera87, 2023-11-18:
OMG I found out what issue is exactly. Everytime when you mess around with the display settings from the Miyoo Mini and don't reboot the device after you change some settings, emulation is stuttering like crazy. After rebooting the device, the stuttering issue is gone! Not sure how I did not notice this before but maybe I was messing around with the contrast settings alot because the newer V4 firmware made some changes to the brightnes and contrast.

closed 2023-11-18 by Fendera87