OnionUI/Onion #1283: GameSwitcher freezes after connecting to wifi

XK9274, 2023-11-17:
### Discussed in https://github.com/OnionUI/Onion/discussions/1282

<div type='discussions-op-text'>

<sup>Originally posted by **whitehollow** November 17, 2023</sup>
Hi dev team, can we please fix this issue where after connecting to wifi on miyoo mini plus and click the menu button to start GameSwitcher it goes to black screen?

If i click the power button for it to go to standby and then click again i can see the screenshot of the last played game but you cannot do anything afterwards.

The only workaround is to restart the device.

Thank you</div>

XK9274, 2023-11-17:
one avenue for a potential fix for this is below however it may need converting to C due to how GS is launched. 

when wifi is restarted, udhcpc and wpa_supplicant may be started with libpadsp.so preloaded, this is bad as they can hold mi_ao open even after audioserver has been killed.

```sh
libpadspblocker() { 
    wpa_pid=$(ps -e | grep "[w]pa_supplicant" | awk 'NR==1{print $1}')
    udhcpc_pid=$(ps -e | grep "[u]dhcpc" | awk 'NR==1{print $1}')
    if [ -n "$wpa_pid" ] && [ -n "$udhcpc_pid" ]; then
        if grep -q "libpadsp.so" /proc/$wpa_pid/maps || grep -q "libpadsp.so" /proc/$udhcpc_pid/maps; then
            echo "Network Checker: $wpa_pid(WPA) and $udhcpc_pid(UDHCPC) found preloaded with libpadsp.so"
            unset LD_PRELOAD
            killall -9 wpa_supplicant
            killall -9 udhcpc 
            $miyoodir/app/wpa_supplicant -B -D nl80211 -iwlan0 -c /appconfigs/wpa_supplicant.conf & 
            udhcpc -i wlan0 -s /etc/init.d/udhcpc.script &
            echo "Network Checker: Removing libpadsp.so preload on wpa_supp/udhcpc"
        fi
    fi
}
```

XK9274, 2023-11-17:
yes, the above fixes it, in combination with:

https://github.com/XK9274/framebuffer-purge-miyoo

Note: If the blackscreen happens, the framebuffer properties are now corrupt, you need to restart unless you've saved the good properties using fbpurge

**Steps to reproduce the black screen:**

- Turn on device
- Turn off wifi
- Turn on wifi
- Open gameswitcher, observe black screen
- Restart device.

**Steps to manipulate blackscreen:**

- Turn on device
- Turn off wifi
- Turn on wifi
- run `fbpurge save`
- Open gameswitcher, observe black screen
- kill -9 gameSwitcher
- Run the `libpadspblocker` function from above
- Run `fbpurge restore`
- Open gameswitcher, observe fault fixed.

Confirmed problem as caused by relationship between libpadsp.so, wpa_supplicant and udhcpc.

Probably to fix, we incorporate a new header/c file that does what libpadspblocker does, run that from keymon if we detect libpadsp.so has a lock on wpa_supplicant/udhcpc, it will cause a quick loss of wireless connectivity but will fix black screens. 

The alternative, is to `unset` LD_PRELOAD of libpadsp.so when calling wpa_supplicant and udhcpc to start (which i believe is still via update_networking.sh (check_wifi))

XK9274, 2023-11-18:
Please can you test https://github.com/OnionUI/Onion/commit/c9e1e789ce2ad6d0a15a9446e1d412e337aa8505

It will give a short black screen when the fix needs to be actioned, ie when you enable wifi and open gameswitcher, you'll get a longer than usual black screen. 

If this fixes we will use this as a basis.

XK9274, 2023-12-16:
https://github.com/OnionUI/Onion/issues/845

This is another duplicate of the above.

JakiShen, 2024-01-14:
Add Gamelist.xml file generator please

XK9274, 2024-01-14:
> Add Gamelist.xml file generator please

Stop posting this on every issue/thread. Make a discussion with a feature request.

JakiShen, 2024-01-14:
Okay Ill do that Im new to github dont know how to

XK9274, 2024-01-14:
> Okay Ill do that Im new to github dont know how to
> 

Please delete all the comments you made elsewhere. Go to discussions and raise a new one