OnionUI/Onion #1296: Missing Favourites sorting

Goodingn, 2023-11-29:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2.3

### Miyoo Firmware Version

X

### Provide a clear and concise description of the issue

In v4.1.x there used to be a tweak to sort favourites, this is missing in v4.2.x

### Steps to Reproduce

Missing

### Provide any additional context or information that may help us investigate your issue

Missing

Aemiii91, 2023-11-29:
It's not missing. It has moved to GLO.

closed 2023-11-29 by Aemiii91

Goodingn, 2023-11-29:
What is GLO?

Goodingn, 2023-11-29:
Thanks. Maybe 'Y Options' needs to be on the screen next to 'A select B back'?

Aemiii91, 2023-11-29:
Not possible, frontend is closed source.