OnionUI/Onion #1306: HTTP web-interface: dark overlay remains on the screen after action is taken

milicomm, 2023-12-06:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2.3

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

HTTP web interface
Once loaded, it has a nice contrast and things are nicely visible but as soon as action is taken, the dark overlay turns on.
On Chrome/Edge (based on Chromium) the overlay remains on the page even once the action is completed.
The only workaround is to refresh the page.

### Steps to Reproduce

Start HTTP web interface in web browser
[image]

Upload file/take any action.
[image]

Page remains dark

### Provide any additional context or information that may help us investigate your issue

No additional comments

XK9274, 2023-12-06:
Hi there.

This is third party source. 

Could you take a look at the repo https://github.com/filebrowser/filebrowser and see if this is reported/resolved there since our version? 

If it's a known issue at source and has been fixed then we'll rebuild it & update our version.

milicomm, 2023-12-06:
> Hi there.
> 
> This is third party source.
> 
> Could you take a look at the repo https://github.com/filebrowser/filebrowser and see if this is reported/resolved there since our version?
> 
> If it's a known issue at source and has been fixed then we'll rebuild it & update our version.

Thank you
Was not aware that it is a 3rd party tool
Checked their issue log and managed to find a reported issue:

https://github.com/filebrowser/filebrowser/pull/2582

Seems that they have committed the change of the CSS files:

https://github.com/langren1353/filebrowser-player/commit/028c0ee3b30845d3324ae6fa6b6c51e57b5feb16

XK9274, 2023-12-06:
No problem at all.

Looking at their releases they provide ARMv7 dists here https://github.com/filebrowser/filebrowser/releases/tag/v2.26.0

You could test this in place of our binary which is https://github.com/OnionUI/Onion/tree/main/static/build/.tmp_update/bin

If you feel this fixes the problem and doesn't introduce any other bugs, you can PR it to contribute or let me know and i'll PR it. 

Thanks for your investigations :)

milicomm, 2023-12-06:
Hi [XK9274](https://github.com/XK9274)
Unfortunately, I am not a developer/programmer.
Using your bin or performing push/pull requests are things which I am not familiar with (occasionally I use Visual Studio and had it even somehow connected to some repos but that's where my knowledge ends).
Sorry :(

milicomm, 2023-12-06:
Great! Allow me a few minutes. 
Just need to do some work (and maybe a coffee) and will come back to you with the results

XK9274, 2023-12-06:
Sure! no rush at all 👍

milicomm, 2023-12-06:
I wouldn't be surprised if this is not the path that I should follow.
Created a new fork into my github, added pull request when seletec to add the updated filebrowser.
merged and well... mentioned the issue in the description of my pull request.

I probably don't understand what is a fork, branch, pull or push request (at all) :-(

XK9274, 2023-12-06:
Heh, that's ok - it looks like it merged into your own fork there. It would need pushing upstream to us. 

I can create this one and co-author you in the commit if you'd like? Moving forwards I think I can spend some time making a guide that's a little easier to understand for contributing and put that on our website?

milicomm, 2023-12-06:
Yeah. I think I messed it up even more.
Don't mind if you take it from here.
Seems I am just too old to understand the concept and how this works (despite https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/about-pull-requests)
For me, this is just one system which does not make any sense to me at all.
If you never worked with pull/push requests, commits, all of this just does not make a lot of sense.

I am truly sorry.

edmunek, 2023-12-06:
co-author

XK9274, 2023-12-06:
Please don't be embarassed. Thankyou for trying and taking the time to get involved. 

Github can be very confusing, took me a while to get the hang of it - i'll try and put something together to make it easier 

And sure, i'll use [edmunek](https://github.com/edmunek) as the coauth.

milicomm, 2023-12-06:
Thank you. I really appreciate all the understanding and patience.

milicomm, 2023-12-07:
BTW, just noticed, that the newer version offers nice progress info on the uploaded files.
previously it was only showing a progress bar, now it also shows the upload info

[image]

closed 2023-12-11 by schmurtzm