OnionUI/Onion #1319: Build branch 'main'

belmix, 2023-12-12:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2.3

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

Build branch 'main'
tweaks - utility (error)

### Steps to Reproduce

Apps - tweaks - utility (error)

### Provide any additional context or information that may help us investigate your issue

[image]

XK9274, 2023-12-12:
Hey,

We're aware of it - there's still some outstanding PR's to be merged that will fix it. cheers

XK9274, 2023-12-22:
Resolved when NUM_TOOLS was updated.

closed 2023-12-22 by XK9274