OnionUI/Onion #1321: Can't connect to already added Wifi

ffuuuuu, 2023-12-14:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

Onion V4.2.3

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

I upgraded my Miyoo Mini+ Onion from the Beta 4.2.0 and had a Wifi already added. The Wifi is visible in Settings -> WIFI with full signal strength. It just won't connect to it. Even when entering the password manually. I did remove the wpa_supplicant via better-wifi-miyoo and even added the Wifi there, but to no avail. When reentering the password via the settings it says: can't connect / reboot router, which I did. Tried different Miyoo official firmwares the only thing that worked was downgrade back to V4.2.0-beta-41caa40. It connected instantly and stayed connected.

### Steps to Reproduce

I don't know how you could reproduce this. It's a simple Fritzbox setup with WPA2 as encryption, and as I said it works fine in Beta 4.2.0 so something must have changed on the Wifi integration with Onion as it isn't even related to the Miyoo firmware itself.

### Provide any additional context or information that may help us investigate your issue

Maybe it is possible to revert the Wifi changes in Onion OS to the ones in Beta 4.2.0 as it was working there and on the versions before that.

XK9274, 2024-01-13:
Without logs it's really difficult to reproduce this as out of 3 of my MMP none have this issue.

Things to try:

- Reset MainUI through tweaks -> advanced -> reset settings
- Disable NTP
- Run a log snapshot and send us it: Tweaks -> advanced -> diagnostics -> snapshot - find the file in the sdcard called log_export.7z

XK9274, 2024-05-17:
Labelling as above until one of us can reproduce the issue.. So far unable to. Also @ffuuuuu can you confirm this is still an issue for you?

oherau, 2024-05-17:
No special logs to provide from my side but I noticed wifi is more prone to cut-off with the megadrive/genesis emulator.

ffuuuuu, 2024-05-18:
Latest update to stable seemed to fix that specific issue. Thank you for your support.

closed 2024-05-18 by ffuuuuu