OnionUI/Onion #1328: Multi profiles & Kid mode

schmurtzm, 2023-12-15:
The idea is to offer multiple profiles and not only Guest profile.
Ideally be able to create a new profile with a name.

The ability to specify a profile type could be nice : Normal / Limited profile.
A limited profile will have only access to Consoles section (may be to video player too ?) 
We also need a way to exit limited profile (combo key ? a fake emulator to run a script with a password to enter ? ...)

It will impact  : 
>    - theme
>    - favorites
>    - recent

and all the stuffs in `Saves` folder : 

>    - Saves
>    - States
>    - Video filter and scaling options
>    - Core options
>    - Custom button remapping layouts
>    - RetroArch history
>  
>    The Game Switcher list will be different to reflect this:
>    - RetroArch playlists and favorites
>    - Timers for your games Play Activity app
>    - RGUI config RetroArch Graphic User Interface

jesseleite, 2024-01-24:
I would SOOOOO love this feature as I have multiple kids that I share my Miyoo Mini Plus with ❤️

A few requests...

- Option to show dedicated 'Profiles' switcher icon on left of home screen?
  - Just like how you can enable Recent Activity app icon on home screen
  - This would make it feel more similar to Playstation menu, how you can select profile from left-most icon on home screen
- Ability to configure dedicated RetroAchievements accounts per profile?
  - Right now I have to switch to guest mode for my kids, AND disable wifi to prevent RetroAchievements from logging into my account while they're playing
  - I realize you can configure RetroArch config overrides, but maybe this could be easier / more automatic with multi profiles?

Sichroteph, 2025-09-22:
The vision I had for this implementation was, on one hand, to allow the creation of profiles with custom avatars.
For the child mode, it would have been a console locked to a single game, even after restarting. Unlocking this mode would require a long press or a key combination, similar to the "app pinning" mode on Android or the automatic bootstrap startup on a jailbroken DSi.