OnionUI/Onion #1349: Miyoo Mini + speaker buzz on screen brightness < 10

sburrows87, 2023-12-30:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

If the screen backlight brightness is anything less than 10 (maximum), then an audible buzz can be heard from the speaker. It appears to be related to the backlight PWM control frequency. Can this be modified or changed through the device tree or similar? It's quite annoying, and doesn't seem to be as prominent on the original Miyoo software.

### Steps to Reproduce

Set screen backlight brightness to anything less than maximum (10).

### Provide any additional context or information that may help us investigate your issue

I am not technical enough to be able to view and modify the linux device tree myself to determine whether the backlight PWM frequency can be changed from 1000Hz (or similar) to a higher value (less likely to be heard).

tGecko, 2023-12-30:
This feature is coming in 4.3 #1179

closed 2023-12-30 by Aemiii91

sburrows87, 2023-12-30:
That's great, is there a beta available for me to try yet please? Thanks again