OnionUI/Onion #1363: Previously installing DraStic then updating to V4.3.0 beta causes issues

dreams897, 2024-01-08:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.3.0 Beta

### Miyoo Firmware Version

202305052130

### Provide a clear and concise description of the issue

I had previously installed the latest version of DraStic on Onion V4.2.3 and it was working fine. Just updated to V4.3.0 beta via the OTA update Menu on device and it installed fine but now DraStic is missing from the emulator list and I see it put my old version in a zip folder called "DraStic-old.zip". How do I get it back? I am also unable to open the package manager, it just hangs when I try to open it. 

### Steps to Reproduce

Installed DraStic manually on Onion V4.2.3 --> Update via Onion OTA update to beta version 4.3.0 --> Reboot device and see that DraStic is now missing from emulator list and package manager does not open

### Provide any additional context or information that may help us investigate your issue

N/A

dreams897, 2024-01-08:
I think I figured it out, the icon and name changed to NDS, I think it was something different before? Still not sure on what happened to the package manager but I ended up doing a clean install

schmurtzm, 2024-01-10:
Hi, the old Drastic install is migrated to Onion during the update. This is important to support the new shortcuts, the save states and the Game Switcher.
The Emu/NDS folder contains Drastic to fit with the convention of other emulators of Onion.
We have also worked with Steward to migrate NDS saves and savestates in the "Saves" folder of Onion which allows you to backup your data as any Onion emulator.

About your Package Manager, it looks like a corruption of data on your SD card. My advice would be to put your SD card in a computer, make a scandisk on it, then extract the archive of the last Onion on it. At boot your Miyoo will update again and everything should be fine.