OnionUI/Onion #1366: Civilization Revolution Freezes upon launching in drastic

codsand, 2024-01-09:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

V4.3.0

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

Launching civilization revolution in drastic results in the device freezing with a white screen with the resolution text frozen on the display as well. Pushing the button for game switcher makes the device vibrate but does not give you access to game switcher or the option to exit drastic with a long press. This game used to work with with v4.2.3 with drastic loaded onto the device but stopped working with the beta update. 

### Steps to Reproduce

Just run civilization revolution in drastic. Used to work on previous v4.2.3 with drastic loaded onto device.

### Provide any additional context or information that may help us investigate your issue

The device just freezes with white screen and resolution being displayed and cannot exit drastic via the center button for game switcher and must be restarted.

heydemoura, 2024-01-19:
This might be related to #1367 .  As the outcome is the same, could be there's a save state corrupted on your device. Please try [my workaround](https://github.com/OnionUI/Onion/issues/1367#issuecomment-1899511677)

codsand, 2024-01-20:
Yes, the issue seems to be the same, I checked the save files using file explorer

closed 2024-06-23 by Aemiii91