OnionUI/Onion #1367: OnionOS 4.3 Beta: Drastic is saving multiples of .DSS save files, and causing save corruption.

laundr, 2024-01-11:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

v4.3.0-8ae761fc

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

Playing Pokemon Platinum on Drastic emulator on Miyoo Mini Plus.

After a few gaming sessions, the game fails to boot. It has happened twice, one of them I noticed it happened after I turned off the screen and then allowed the device to sleep on its own after a 15 minute wait in sleep mode.

When starting the device the game boots to a white screen and then bounces me back to the main menu. Attempting to open the game again does the same thing. 

Pulling the SD card and inspecting the save files on my computer reveals that there are 2 .DSS and one .DSV save file for the game. 

The file titles are:
Pokemon - Platinum Version (USA) (Rev 1).dsv
Pokemon - Platinum Version (USA) (Rev 1)_0.dss
Pokemon - Platinum Version (USA) (Rev 1)_10.dss

### Steps to Reproduce

Start a game of pokemon platinum
Save the game
Single-press the power button to turn off the screen
Allow the device to power itself off after the sleep period
Boot the device and see if the game will run.

### Provide any additional context or information that may help us investigate your issue

Upon inspecting the save files in my "NDS_OLD.7z" folder, I see that multiple saves were not created for this game. Only a single .DSS file with the title: "Pokemon - Platinum Version (USA) (Rev 1)_10.dss"

Ulquiro, 2024-01-13:
I encountered this issue with all the NDS games i have tried so far, all of them with the same steps to reproduce

heydemoura, 2024-01-19:
I faced the same issue with Pokemon Soul Silver, I believe it might happen to any NDS game. 
The workaround for me was to:

- Insert the SD card on the computer and navigate to the NDS states folder. ( Saves > CurrentProfile > states > Drastic )
- Rename "Game Title_10.dss" to "bkp.Game Title_10.dss"
- Rename "Game Title_0.dss" to "Game Title_10.dss"

The last save I had was restored. 

Maybe Drastic saves are not yet being well handled when turning off the device with sleep mode.

codsand, 2024-01-20:
Yes, the save file is saving wrong from sleep

BaseInfinity, 2024-07-13:
Is there a way to disable the default to autosave when powering off? I've run into this myself and I'd rather just disable the feature.

BaseInfinity, 2024-07-26:
https://github.com/OnionUI/Onion/blob/main/static/packages/Emu/Nintendo%20-%20DS%20(Drastic)/Emu/NDS/resources/settings.json#L23-L24
Looks like this is probably due to the `auto_slot` value here that is set to 10.

Should we just set it to `-1` or `1`? @Aemiii91

Aemiii91, 2024-07-26:
@BaseInfinity I believe `-1` is an invalid value, and `0-9` is reserved for manual save states

dominat0r91, 2024-09-03:
The same issue occurred to me today after playing Pokémon Platinum via Drastic the night before. That time the game also auto-saved after the device switched off via standby mode. The solution from @heydemoura worked for me as well, thanks for that!

What would be a long-term solution though to prevent this from happening again? Wouldn't this issue bug all users who regularly play NDS via Drastic?

BaseInfinity, 2024-09-04:
@dominat0r91 something easy would be to disable autosave on sleep so it at least prevents saving the wrong slot.

AFAIK this problem does not happen when you save state yourself

TwanoO67, 2024-10-31:
I encoutered the same issue on "Pokemon - perle".
( Single-pressed the power button to turn off the screen > Allow the device to power itself off after the sleep period)

OnionOS is crashing on start (I had to remove .tmp_update/cmd_to_run.sh, to be able to boot again).
But when retrying to start the game, retroarch is crashing, and I have to delete file on sd and reboot again.

Now the only file I have in states is:
"Pokemon-perle_10.dss"

Is there a way to recover the game from that ?

Biscuits25, 2024-11-30:
This sounds like the same thing as issue #1406 https://github.com/OnionUI/Onion/issues/1406

miguelbaldi, 2024-12-04:
I've found this issue on the Drastic repository:
https://github.com/steward-fu/nds/issues/48
I think it's related

BaseInfinity, 2024-12-04:
Great! I'm glad it was simple!

Biscuits25, 2024-12-05:
I plugged in my onion sd card and found the drastic settings.json file and edited the autoslot to be 0. Im still getting drastic freezing up when i boot.

I put it to sleep while playing a game and then wait for it to fully shutdown, then when i boot it up it will be frozen on this screen. 
[image]

Its not every time, it took about 5 boots for it to freeze up. After deleting the drastic savestate it boots fine. I did a second test and it only took 2 boots for it to freeze again.