OnionUI/Onion #1373: PICO-8 (standalone) exit button is not working

worthis, 2024-01-13:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.3.0-beta

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

Pressing the Menu button does not close the PICO-8 (standalone).
The only option is to turn off the console by long pressing the Power Off button.

### Steps to Reproduce

1 Download "The Lair" cartridge [https://www.lexaloffle.com/bbs/cposts/th/thelair-0.p8.png](https://www.lexaloffle.com/bbs/cposts/th/thelair-0.p8.png).
2 Place it to the "Roms/PICO" folder.
3 Run it by PICO-8 (standalone) from the Expert menu.
4 Try to close it by pressing the Menu button.

### Provide any additional context or information that may help us investigate your issue

PICO-8 (standalone)

XK9274, 2024-01-13:
Works fine here following your steps. 

Did you have the App version of pico8 and copy anything over by any chance? If you moved any dirs like: `lib` you've overwritten the Onion version of the libs, and your exit button will now be `Menu + Select`

https://github.com/OnionUI/Onion/assets/47260768/d52ab535-2f8f-43d0-ad92-a83571d6438b

worthis, 2024-01-13:
Strange, pressing SELECT + MENU brings up the standard RetroArch menu.

XK9274, 2024-01-13:
You must somehow be running with fake-08 through RA. 🤔

worthis, 2024-01-13:
https://github.com/OnionUI/Onion/assets/47685963/c1b3a64c-db37-498a-b5ca-3e4117e9f1b1

[image]

XK9274, 2024-01-13:
https://github.com/OnionUI/Onion/assets/47260768/749b80b5-9fb2-41f5-8658-46e9b3d476c4

Still works with that cart 

[image]

Can you run `ps` on ssh or telnet and see if it says Fake08, or Retroarch or something else?

worthis, 2024-01-13:
Problem solved.
There was a cfg file for this cartridge in the Roms/PICO folder, I deleted it.

closed 2024-01-13 by worthis

XK9274, 2024-01-13:
Thanks for the update :)