OnionUI/Onion #1383: Set time automatically causes crashes

rossmessing, 2024-01-18:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.3.0-beta

### Miyoo Firmware Version

202305052130

### Provide a clear and concise description of the issue

After enabling automatic time setting, I observed frequent crashes, particularly after opening PSX / DS games.  Loading screen would spin until I powered the device off.  Powering the device on would load the game that had been loading.

This problem disappears with automatic time setting disabled.

I've only seen this on 4.3, and only on wifi.

I did reinstall onion and verify that this still occurs

### Steps to Reproduce

* install onion 4.3
* set date and time setting to automatic via internet
* Run a PSX or DS game (says loading indefinitely)
* Restart, and the game runs
* Run a game with any system, loading indefinitely keeps happening.

### Provide any additional context or information that may help us investigate your issue

I don't really need the automatic time setting functionality, but it seemed like an issue, and surprised me, so figured I'd write up the bug report.

XK9274, 2024-01-18:
If you're getting stuck at "Loading", please try the two files from this PR https://github.com/OnionUI/Onion/pull/1369 and report back if it fixes it for you.

XK9274, 2024-02-27:
@rossmessing Can you confirm if you still have this issue on 4.3 stable? It should be fixed.

closed 2024-06-23 by Aemiii91