OnionUI/Onion #1385: Select button ,(to refresh roms) does not work after using File Explorer.

bteamfox, 2024-01-20:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2.3 and 4.3 Beta

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

Using File Explorer to rename Roms causes the UI option for refresh Roms not to appear when select is pressed.

### Steps to Reproduce

Copy new games to roms/commodore/filename.d64 using FTP
use select button to refresh roms. 
check game runs close game using select/menu, back,quit retroarch.
Start File Explorer,
Rename some Rom names e.g. roms/commodore/filename.d64 to newfilename.d64 - repeat this two or three times
Press Menu button
Select Quit
go back to main UI front.
Press Select to bring up Refresh Roms option, but most of the time it will not appear.

If you start quick guide, then go back to main UI, then press select, the Refresh roms option will now appear.

### Provide any additional context or information that may help us investigate your issue

I copy most of my commodore c64 files to miyoo mini plus using ftp. I copy the .d64 files to the roms/commodore folder.
I then check the game runs ok on the MM+ once the games has run I then use the File Explorer to rename the file or files.
I then use Select button to refresh roms. Which does not then work.

tGecko, 2024-03-11:
I've tried multiple times, but cannot reproduce this.

Can you confirm these are the exact steps you're taking? In the beginning of the video I transfer `nightyknight.d64` to `Roms/Commodore` via FTP
https://github.com/OnionUI/Onion/assets/6023076/d7af23e2-daf9-491b-9afb-f876d1a2fe9c

closed 2024-06-23 by Aemiii91

Evergreen85, 2024-10-19:
I just had this issue. Just used file explorer and added a bunch of roms. Refreshed roms 1 time and it worked. Now it wont work at all. Google brought me here...