OnionUI/Onion #1390: Unable to download OTA through 4.3.0...e33 to 4.3.0...326

Amiga500, 2024-01-23:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.3.0-RC-6e310e33

### Miyoo Firmware Version

202105052130

### Provide a clear and concise description of the issue

Trying to update via OTA from 4.3.0-RC-6e310e33 to latest one (75c2b326) is not possible to download latest version. Pressing A t the screen come back to App list page
[image]

### Steps to Reproduce

Please provide detailed steps that will help us replicate and investigate your issue

### Provide any additional context or information that may help us investigate your issue

no

Aemiii91, 2024-01-24:
I'm very sorry, but a bug slipped through to the OTA update script. If you updated to `6e310e33`, you'll either need to install Onion manually (via file transfer), or replace `.tmp_update/script/ota_update.sh` with the [latest one](https://github.com/OnionUI/Onion/blob/main/static/build/.tmp_update/script/ota_update.sh)

closed 2024-01-24 by Aemiii91

Aemiii91, 2024-01-24:
If you're completely out of luck and can't use manual file transfer, then a workaround could be to fix the `ota_update.sh` script in the on-device File Explorer.

Simply replace `Release_size_bytes` with `Release_size` (case-sensitive) as seen on line 132 and 133 below:

[image]

In File Explorer (`.tmp_update/script/ota_update.sh`):

[image]

Amiga500, 2024-01-24:
Thanks for the explanation Aemiii91, i had resolved with two manual updates. 
Keep up with this fantastic work on Onion!