OnionUI/Onion #1392: Onion OTA Updater Announces Update Available But Closes When I Choose To Update

jgarciabu, 2024-01-24:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.3.0-RC-75c2b326

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

I enter the onion ota updater and it tells me there’s an update available on the beta release option (4.3.0-RC-b66469de). It says press a to update. When I press a it just exits the updater and takes me back to apps section of onion os where I launched the ota updater from

### Steps to Reproduce

1. Enter the Onion OTA Updater 
2. Click the beta release option
3. It tells you there’s an update available and to press a if you’d like to update
4. Press a
5. It closes out of the updater and dumps you out to the apps section of onion OS 

### Provide any additional context or information that may help us investigate your issue

N/A

tGecko, 2024-01-24:
Duplicate of #1390 - Please update manually

closed 2024-01-24 by tGecko

jgarciabu, 2024-01-24:
Thanks for the link to the duplicate issue. The proposed fix worked. Apologies for the duplicate issue. I love all the work you all are doing on Onion OS!