OnionUI/Onion #1393: ScummVM Standalone high pitched audio

dreams897, 2024-01-25:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

Onion V4.3.0-RC

### Miyoo Firmware Version

202305052130

### Provide a clear and concise description of the issue

ScummVM standalone still has high pitched audio as of update Onion V4.3.0-RC. I'm wondering if there is a way I can underclock the Miyoo Plus or something to help fix this issue? It just seems to be running too fast or something. The music audio sounds perfect it's just the dialogue audio in games like Broken Sword 2.5 and the Spyfox games for example are very high pitched. The issue doesn't exist in the core ScummVM so I don't know why it's happening in the standalone when it's a newer version. Any sort of help would be appreciated, but maybe there is nothing that can be done for now

### Steps to Reproduce

Launch a game such as Broken Sword 2.5 in ScummVM standalone app and experienced high pitched dialogue audio

### Provide any additional context or information that may help us investigate your issue

N/A

dreams897, 2024-01-25:
I also underclocked and overclocked to see if that would fix it but it's exactly the same. Seems to be audio frequency related and I'm not sure how to change that

Aemiii91, 2024-01-25:
We already have an open issue about this (#786), that you also commented on. Closing this as duplicate.