OnionUI/Onion #1395: Game slow to start back up from sleep mode

bhill89, 2024-01-25:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.3-RC

### Miyoo Firmware Version

202306111426

### Provide a clear and concise description of the issue

When putting the game to sleep and coming back to it after a decent amount of time (haven't actually timed it), the game appears to be frozen, but eventually starts up again when waiting a minute.

### Steps to Reproduce

Currently playing Pokemon HeartGold, put the game to sleep and do something like making a sandwich or taking a shower, then wake up the device. Device wakes up, but game is frozen for longer than a few seconds.

### Provide any additional context or information that may help us investigate your issue

I just wanted to mention it. Not a big deal to me. Everything else has been working great!

Aemiii91, 2024-01-25:
What do you mean by "to sleep" and "wake up" ? Powering off, or simply having the screen off (standby) ?

bhill89, 2024-01-25:
Standby

DanCousins, 2024-02-22:
I'm experiencing something similar. Very slow to load a game, extremely slow when trying to turn volume up or down, or adjust screen brightness. Everything has several seconds of delay.

It's not consistently bad, as I tried to time my boot into a game just now, and found it was back to normal. Will update if it happens again, but has happened several times since updating to the latest beta.

DanCousins, 2024-02-24:
Just experienced it now, volume up/down and brightness up/down very slow to respond. Wasn't paying attention to the boot time. 

Thought I would install the terminal to check the CPU usage to see if anything was maxing it out, but the usage was essentially 0, although opening up the package manager to enable terminal took a long time - probably 45 seconds or so.

Games seem to run alright, and after a reboot it's back to normal, but does pop up again from time to time.