OnionUI/Onion #1398: Mini v4 shows 640x480 as scaling size when auto-resuming one game, 560p when launching from menu

theturtlemafia, 2024-01-28:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.3.0 RC

### Miyoo Firmware Version

202310271401

### Provide a clear and concise description of the issue

I have a Miyoo Mini V4 and updated from 4.2.3 to 4.3.0-RC. When I use auto-resume for a certain game, the scaling acts as if the screen is 640x480. Integer scaling is small, and the overlay text when switching with the hotkey will show 640x480. If I open the game from the "consoles" menu, or from the game-switcher, integer scaling is correct (larger) and the overlay text shows 752x560.

It only happens for me in Pokémon Gaia, which is an unofficial GBA romhack. But it was also the game I had active before updating to 4.3.0.

It could just be a problem with the romhack, but my concern is this might instead be a problem with the previous game someone has played before updating to 4.3.0

### Steps to Reproduce

Turn off the miyoo while playing Gaia and then turning it back on, auto-resuming Gaia. It will show 640x480 and integer scaling modes are small until I close the game and re-open it.

If I use auto-resume with any other game (I have tested a variety of games and consoles), or if I open Pokemon Gaia from the consoles menu or from the game-switcher the scaling text shows 560p and integer scaling is correct. It is only when opening Gaia from auto-resume.

Instead of Gaia, the reproducer may have to try this with the game they last played before updating to 4.3.0

### Provide any additional context or information that may help us investigate your issue

Miyoo Mini v4 model my283

I was running the current Stable release before updating to 4.3.0 RC.  Before updating, I shut down my Miyoo Mini v4 while playing Pokemon Gaia, the game that causes the scaling bug for me.

tGecko, 2024-02-15:
I'm able to reproduce this issue with this particular game, on a fresh 4.3 RC install, so it's not an upgrade issue.
Currently trying to fix it. As a workaround, it doesn't happen with gpSP, so you can change the core to that.

theturtlemafia, 2024-02-15:
Hi, thanks for looking into this! I'll switch to gbSP for that game. Glad to hear it's not a general update issue.

commit 61ff355a (2024-02-24): Create and enable swap partition on boot. (#1428)

Normally, this is done by MainUI, but when we auto-resume a game, MainUI
doesn't get a chance to do it.
MainUI checks if the swap exists already, so no double swapon happens.

Fixes #1398 and maybe some other problems that went under the radar..

Negative impact:

If the cachefile exists (which is the norm) only swapon runs, which is
near instant.
If the cachefile doesn't exist, it takes a while to generate the file on
SD.
```
(runtime) 1970-01-01 00:00:03: Creating swap file
128+0 records in
128+0 records out
134217728 bytes (128.0MB) copied, [... cut]