OnionUI/Onion #1401: FFPlay Unicode characters not displayed properly in file selector

goooooouwa, 2024-01-31:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2.3

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

FFPlay: Unicode characters not displayed properly in file selector 

[image]

### Steps to Reproduce

Please provide detailed steps that will help us replicate and investigate your issue:

Steps to Reproduce:
1. Upload a video with filename containing Unicode characters, e.g. Chinese, Japanese
2. Open FFPlay to select the file

Expected:
- The filename containing Unicode characters can be displayed properly

Actual:
- The filename containing Unicode characters is not displayed properly (see attached screenshot)

### Provide any additional context or information that may help us investigate your issue

I believe the file selector is provided by the SDL terminal, relavant code snippet found in `static\packages\App\Video Player (FFplay)\App\FFplay\launch.sh`:

`./bin/st -q -e "$sysdir/script/file_selector.sh" "Video Player" "$mediadir" "3g2 3gp asf avi flv h264 m2t m2ts m4v mkv mod mov mp4 mpg tod vob webm wmv"`

XK9274, 2024-05-08:
Pretty sure this is a problem with the terminal app's (`st`) bitmap font and not something we can easily fix as I believe eggs was involved. It's not a standard font file and therefore probably doesn't contain the characters you're expecting in the format you're expecting. 

You can see the same issue here where the progress bar should be.. well a bar - not p's

[image]

Esn024, 2024-08-30:
In the File Explorer, though, there is no such issue and unicode filenames display just fine.

fengfrw, 2024-08-30:
It is recommended that FFpay\PDF\PixelReader be moved down to support Unicode character display and convenient and fast list selection.