OnionUI/Onion #1404: easyRPG creates one entry in the Activity Tracker per initation

jujemido, 2024-02-02:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.2

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

Hello!

I'm playing Yume Nikki, and easyRPG game, and every time that I open the game, it creates one entry in the Activity Tracker.

### Steps to Reproduce

Opening an easyrpg and playing one game.

### Provide any additional context or information that may help us investigate your issue

Is it possible to merge entries? Even manually
[image]

jujemido, 2024-02-02:
Could someone help me to remove all these entries/merging them if I give you access to my play activity sqlite? 🥺

Prozi, 2024-02-11:
> Could someone help me to remove all these entries/merging them if I give you access to my play activity sqlite? 🥺

image if this db would be a simple JSON file

Prozi, 2024-02-26:
> > Could someone help me to remove all these entries/merging them if I give you access to my play activity sqlite? 🥺
> 
> image if this db would be a simple JSON file

or actually

imagine activity tracker as a folder of `ActivityTracker/`

with one `.JSON` file per entry, easy to edit, or to delete in any text editor

@schmurtzm @Aemiii91

tGecko, 2024-03-11:
> Could someone help me to remove all these entries/merging them if I give you access to my play activity sqlite? 🥺

Can you provide us your `SD/Saves/CurrentProfile/play_activity/play_activity_db.sqlite` file please

tGecko, 2024-03-11:
Can't reproduce this with the version of the game i'm using.. What version are you using? 
[image]