OnionUI/Onion #1406: Game Switcher freezes after booting up when the first entry is from DraStic

gabrielsoldani, 2024-02-03:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.3.0-RC b66469d

### Miyoo Firmware Version

202305052130

### Provide a clear and concise description of the issue

If Game Switcher is set to be launched on boot, and the last played game is from DraStic, it shows the game and freezes. You have to reboot the device to continue. You may get stuck in a loop.

### Steps to Reproduce

Make sure Game Switcher is set to open on boot. This was the default when I first installed Onion, but you may have changed it.

1. Start a DraStic game.
2. Shut down the device by holding down the Power button for a few seconds. This should show the usual "saving and shutting down" screen.
3. Power up the device. After the Miyoo logo, it should show the Onion logo and "Waiting for network", as usual. Note: I was in range of my Wi-Fi network. Not sure if this is required to reproduce.
5. Game Switcher launches. You can see the first entry, with the game you were last playing and a screen capture of the save state you created in step 2.
6. Try to press any button. Game Switcher is frozen.

In order to continue, you must shut down the device forcefully by holding down the Power button until it turns off. After powering up the device, it may either repeat steps 3-6 (at which point you should try again) or it may launch Game Switcher, but with a different first entry (in my case it was the game I was playing previously, Celeste, on Fake-08).

### Provide any additional context or information that may help us investigate your issue

I was playing Phoenix Wright - Ace Attorney (USA), decrypted, matches No-Intro, but it should happen to any DraStic game.

I was in range of my Wi-Fi network when booting up. Not sure if it makes any difference.

You may work around the issue by tweaking the settings and configuring MainUI to launch on boot instead.

First reported and corroborated on [this /r/MiyooMini thread](https://www.reddit.com/r/MiyooMini/comments/1ae15t4/issues_with_drastic_saves_on_onion_os/kk7k4jo/).

May be related to #1350 and #1283.

gabrielsoldani, 2024-02-03:
Since users may get stuck on boot, and one of the main features of 4.3.0 is DraStic support, I believe this should be investigated before the general availability of 4.3.0.

I'd be happy to provide more details and test any patches.

Sichroteph, 2024-02-04:
Thanks for your feedback, however I can't reproduce this bug following the given steps on my miyoo mini. 
- Is it consistent with any drastic game ? 
- can you describe what is ticked in the startup page of tweaks ? 
I can also try to five you an early build and see if that fixes it for you.

gabrielsoldani, 2024-02-05:
@Sichroteph

>I can't reproduce this bug following the given steps on my miyoo mini.

I'm on the Miyoo Mini+, if that helps. It does not seem to happen every time. I'll go back to using GameSwitcher as my "Start application" for the next few days to see if I can gather more information and keep this thread updated.

>Is it consistent with any drastic game ?

I'm not sure. It only happened to me with this game. Since a different user on Reddit ran into the same bug, I find it unlikely that we were playing the same game, but that's possible.

>can you describe what is ticked in the startup page of tweaks ?

Sure. I've also included network services and date/time settings as they also influence things that can start on boot.

## System

### Startup

| | |
| :--- | ---: |
| Auto-resume last game | OFF |
| Start application | GameSwitcher |

### Date and time

| | |
| :--- | ---: |
| Set automatically via internet | ON |
| Wait for sync on startup | OFF |
| Get timezone via IP address | ON |

## Network

| | |
| :--- | ---: |
| Samba | ON (everything except SDCARD) |
| HTTP | OFF |
| SSH | OFF |
| FTP | OFF |
| Telnet | OFF |
| VNC | OFF |
| Disable services in game | ON |

Sichroteph, 2024-02-07:
Did not manage to reproduce it on my MMP following your exact settings.
Can you reproduce the issue ? 
What game were you using ? (I believe the exact rom name is important as it can be the culprit here)

Biscuits25, 2024-02-18:
I think I might be the other person on reddit that is also having this issue. I have a regular mini so I dont think its a mmp issue. I tried flashing to a different sd card but that did not help. I have experienced it on Mario Kart DS (USA, Australia) (En,Fr,De,Es,It).nds, and a pokemon platinum romhack. Im trying to test other games but my Miyoo is stuck in the frozen loop and I am having trouble getting it out. Will update if I get more info.

Biscuits25, 2024-02-18:
Well I was able to get out of the boot loop by deleting my drastic save states using my pc, so it definitely seems to be an issue with it loading the save state.

Biscuits25, 2024-02-18:
Just also reproduced it with Dragon Quest IX - Sentinels of the Starry Skies (USA) (En,Fr,Es).nds.

And it doesn't happen every time I had to let it shut down and boot it 4 or 5 times before it froze.

gabrielsoldani, 2024-02-18:
I haven't been able to reproduce it again in the past few days.

@Biscuits25 If you do run into a loop again, could you upload the broken save states?

Sichroteph, 2024-02-18:
I am confused, because the gameswitcher does not manage saves at all. If the gameswitcher is freezing, it can't be because of a save file.

gabrielsoldani, 2024-02-18:
@Sichroteph 
> I am confused, because the gameswitcher does not manage saves at all. If the gameswitcher is freezing, it can't be because of a save file. 

Right. @Biscuits25, any chance you've also deleted a `romScreen`?

Biscuits25, 2024-02-18:
[Dragon Quest IX - Sentinels of the Starry Skies (USA) (En,Fr,Es)_10.zip](https://github.com/OnionUI/Onion/files/14324677/Dragon.Quest.IX.-.Sentinels.of.the.Starry.Skies.USA.En.Fr.Es._10.zip)

> @Sichroteph
> 
> > I am confused, because the gameswitcher does not manage saves at all. If the gameswitcher is freezing, it can't be because of a save file.
> 
> Right. @Biscuits25, any chance you've also deleted a `romScreen`?

No I just tested and I only need to delete the number 10 save state to fix the boot freeze. I've uploaded the most recent one but I can upload others if needed.

> I am confused, because the gameswitcher does not manage saves at all. If the gameswitcher is freezing, it can't be because of a save file.

I don't think the gameswitcher itself is freezing, Drastic is freezing when loading the autosave. I think theres an issue with Drastic making auto saves while the device is in sleep mode.

gabrielsoldani, 2024-02-18:
@Biscuits25 So in your case the device only freezes when you attempt to launch the game from GameSwitcher?

When I ran into this, I couldn't even change to a title in GameSwitcher.

Can you try this: after turning the device off while DraStic was running, turn the device on and when GameSwitcher appears, press the arrow keys to go to a different game, then go back to the DraStic game, and press A. If it freezes by this point, we can rule out most of the Game Switcher initialization code.

Biscuits25, 2024-02-18:
I think we might be having different issues. I just checked and I actually don't have gameswitcher set as my start application. I have it set to Auto-resume last game, but my start application is set to the MainUI.

I was confused when you said to try switching to a different game, with Auto-resume you don't get a chance to switch games during boot.

Sichroteph, 2024-02-19:
> [Dragon Quest IX - Sentinels of the Starry Skies (USA) (En,Fr,Es)_10.zip](https://github.com/OnionUI/Onion/files/14324677/Dragon.Quest.IX.-.Sentinels.of.the.Starry.Skies.USA.En.Fr.Es._10.zip)
> 
> > @Sichroteph
> > Right. @Biscuits25, any chance you've also deleted a `romScreen`?
> 
> No I just tested and I only need to delete the number 10 save state to fix the boot freeze. I've uploaded the most recent one but I can upload others if needed.
> 
> > I am confused, because the gameswitcher does not manage saves at all. If the gameswitcher is freezing, it can't be because of a save file.
> 
> I don't think the gameswitcher itself is freezing, Drastic is freezing when loading the autosave. I think theres an issue with Drastic making auto saves while the device is in sleep mode.

This one hould be fixed it the 4.3 RC 👍

gabrielsoldani, 2024-02-22:
So, since @Biscuits25 said he was not using Game Switcher, but instead "Auto-resume last game", I went ahead and set my device to "Auto-resume last game" as well. I tried turning it off in the middle of Mario Kart DS (USA, decrypted, matches Redump) and then turning it back on for a few times, but nothing abnormal happened. Then after a couple of days without playing I turned it on again and lo and behold, I booted into a frozen Mario Kart DS.

I suppose what could have happened earlier was that I did press A to launch DraStic from Game Switcher before it froze. Or maybe what causes the device to freeze is something that DraStic does on shut down, not on start up. Not sure.

I've updated the thread with this new information.

Sichroteph, 2024-02-22:
Thanks a bunch for sharing your thoughts! It looks like the problem might not be the game switcher after all, but possibly due to state corruption or instability. We'll definitely keep an eye out for similar game switcher freezing situations, although yours is the only one reported so far. Your detailed reports have been super helpful, and if you've got any more insights to share, we're all ears. 
If you feel like we can close the case for now, just give us a heads up. In any case we're here to help!

matthewclaussen, 2024-02-26:
I've had this happen a few times. It seems to happen when I let the device auto shutdown with a NDS game paused in the background.

Sichroteph, 2024-02-26:
> I've had this happen a few times. It seems to happen when I let the device auto shutdown with a NDS game paused in the background.

This issue should be fixed in the last 3 versions 👍

matthewclaussen, 2024-02-26:
I was running 4.3 official when it happened again yesterday. This did happen a few times with the 4.3RC though. Could it be something residual from the RC?

mistercrunch, 2024-03-02:
Having the same problem on a fresh 4.3.0 install, goes straight into Drastic, presumably trying to restore a bad save state and everything there is frozen. Any way to prevent booting into the broken save state or exiting out of it? Some workaround here would be great.

Somehow this happened after I reinstalled (by copying a fresh download on the sd card) because I had another issue beforehand: I would boot in a black screen with 4 small horizontally laid out rectangles on the lower left end of the screen with a `1` next to it (right side).

My next step is to downgrade to 4.2.3 and wait for 4.3.1 I think, unless you have other suggestions.

gabrielsoldani, 2024-03-02:
Had this happen with 4.3.0 stable as well. This probably shouldn't have shipped :/

@mistercrunch Try mashing the Menu button when booting up to exit DraStic before it loads the corrupted save.

mistercrunch, 2024-03-02:
Downgraded to previous stable and things are working. Trying to load that one Drastic bad save state crashes, so I'll go and delete that one from my PC. 

I also change the boot settings to the main UI as opposed to last game, which should prevent this particular issue that effectively soft bricked my device. 

Given this issue and general nature of emulator being unstable, I would suggest changing the default to boot in the main UI. At least as the default, preventing a bad save state from soft bricking the device.

bmn001, 2024-03-23:
Oh man this just happened to me too. Stuck in a soft brick loop. Reboot, Drastic freezes on load, repeat. 4.3.0.

I was able to finally get out of it by holding the menu button on powerup which took me to Game Switcher instead of launching Drastic.

Nate0714, 2024-04-04:
Hello, I had this issue on my MMP and my cousin did as well. The fix is to delete the save state that is corrupted that the game switcher keeps failing to load.  Go into the file explorer app, and navigate to saves/currentprofile/states/drastic and delete all the names of the game that’s frozen. For me that was ff3 and for my cousin that was advanced wars days of ruin. If you save periodically within the games you’ll be fine this just deletes the corrupted game switcher save state it keeps bricking while loading.

Nate0714, 2024-04-04:
If you want to keep playing drastic nds I would disable the auto resume under apps/tweaks/system/startup/ and toggle off auto resume. When I made additional MMP for family/friends I always disable this now since the drastic freeze happens a lot more often when this is enabled.

IcelandicIcecream, 2024-04-23:
> Oh man this just happened to me too. Stuck in a soft brick loop. Reboot, Drastic freezes on load, repeat. 4.3.0.
> 
> I was able to finally get out of it by holding the menu button on powerup which took me to Game Switcher instead of launching Drastic.

This is a bit in the future, but for anyone looking back at this post (4.3.0). You could either stop this soft brick loop by loading the SD card on a PC and then:
1. Deleting the actual corrupted save state from Saves/CurrentProfile/states/Drastic/{game_name}_10.dss. (This is the default save state that's loaded when ever u auto-resume)
2. Navigating to Emu/NDS/resources/settings.json and changing auto_state to 0. (This causes the drastic emulator to reboot every time you start it up - so you lose out on the auto resuming functionality for the game switcher and startup)

v3j3y, 2024-05-31:
Just happened to me. I was playing Mario Kart DS even after disable the auto resume under apps/tweaks/system/startup/ and toggle off auto resume. Now it started happening to almost all games. 
Steps:
1. Go to the game and click power button (once) for screen lock. 
2. Wait for sometime, so the screen lock goes to a state where the power is off
3. Now click on the power button, it boots up again and goes to this frozen screen

nanbomba7, 2024-06-04:
I just got the same thing as the last post, more frecuent than not the console gets bricked when playing a ds game if i power it off and then power it back on. After that i cant play any ds game.

mrchrisster, 2024-07-01:
Same issue for me on 4.3.1-1