OnionUI/Onion #1419: Activity Tracker - Missing Images

Prozi, 2024-02-11:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.3.0-RC-b66469de

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

the activity tracker doesn't show thumbinails for a few PS games that I play for example:
Tony Hawk Pro Skater 4

when I go to favorites/PSX menus I have this thumb scraped

### Steps to Reproduce

desired result: it should show the thumb of the game

actual result: it shows [missing thumbinail] image

possible cause: some games show correctly, so I don't know maybe thats because I reset the activity tracker a few times by deleting its files including db from file manager in onion - that way of reset worked btw

**I think the steps to reproduce is to install THPS4 and then delete activity tracker files and start playing, go quit game and then run activity tracker - you should see no thumb**

### Provide any additional context or information that may help us investigate your issue

bonus question: where does activity tracker link to images or store them? in that nosql db? it would be easier if possible to change to simple json text file

Prozi, 2024-02-11:
this would https://github.com/OnionUI/Onion/pull/1305 possibly fix my issue 👍

Aemiii91, 2024-02-11:
Duplicate of #1221

Aemiii91, 2024-02-11:
@Prozi probably, but it still has some issues that needs sorting. It's not ready.

closed 2024-02-11 by Aemiii91

Prozi, 2024-02-12:
> Duplicate of #1221

@Aemiii91 not sure about that. even if I erase the activity tracker, then scrape images, then the new activity is added - still no image, so IMHO it's a bit different issue

Prozi, 2024-02-12:
> @Prozi probably, but it still has some issues that needs sorting. It's not ready.

can I support this in some way? I can try to do Code Review of that MR with what I know from C++ from before 10+ years 🙈

Prozi, 2024-02-12:
also, thanks for fast reply

AND THANKS FOR HELPING MAKING ONIONOS IT'S GREAT!!!

Aemiii91, 2024-02-12:
I'll try to work on it soon. I'll keep this issue in mind.

Prozi, 2024-02-12:
tyvm