OnionUI/Onion #1440: PSX screen contracted

Vclaiton, 2024-02-25:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.3.0

### Miyoo Firmware Version

202305052130

### Provide a clear and concise description of the issue

After updating to 4.3.0 in some PSX games The screen show a contracted image.

### Steps to Reproduce

Please provide detailed steps that will help us replicate and investigate your issue

### Provide any additional context or information that may help us investigate your issue

X2 game for example.

Aemiii91, 2024-02-25:
What do you mean contracted? Shrinked? Can you attach some screenshots?

Vclaiton, 2024-02-25:
> What do you mean contracted? Shrinked? Can you attach some screenshots?

Yes, shrinked. English is not my language. In some parts of the game the image is ok, in others are shrinked. I will put the images.[image]

[image]

[image]

[image]

Aemiii91, 2024-02-25:
And you're absolutely sure that these games didn't render this way before you updated?

Vclaiton, 2024-02-25:
> And you're absolutely sure that these games didn't render this way before you updated?

Yes, absolutely. I have played they yesterday. Same ROMs, I didn't nothing after playng, and after I updated.

tGecko, 2024-02-25:
Which Formula 1 game is this?

Vclaiton, 2024-02-25:
> Which Formula 1 game is this?

The first one of the Psygnosis for the PSX. 1995 season.

tGecko, 2024-02-25:
After testing I can reproduce this, but I can also reproduce it with Onion 4.2.3. This is an inherent issue with PCX-ReARMed and Playstation games sometimes running in half or full height. 

To fix this, please press Menu + Start to cycle through the display modes, and select "Integer scaling: On (4:3)"

closed 2024-02-25 by tGecko