OnionUI/Onion #1443: Splore stopped working in 4.3.0

TotallyNotSethP, 2024-02-26:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.3.0

### Miyoo Firmware Version

202303262339

### Provide a clear and concise description of the issue

In 4.2, you could run the file "~Run PICO-8 with Splore" via the PICO-8 standalone found in the Expert menu option. In 4.3.0, it gives the error "Could not load /mnt/SDCARD/Roms/PICO/~Run PICO-8 with Splore.pico-8" and boots you to the PICO-8 terminal. Since you can't use the keyboard in PICO-8 standalone, you are unable to do anything...

### Steps to Reproduce

1. Install PICO-8 standalone in 4.2
2. Upgrade to 4.3.0
3. Try to run PICO-8 standalone with Splore

### Provide any additional context or information that may help us investigate your issue

N/A

TotallyNotSethP, 2024-02-26:
I just tried uninstalling and reinstalling the package via package manager and still no luck...

Aemiii91, 2024-02-26:
You need to use the `.png` file: https://github.com/OnionUI/Onion/blob/main/static/packages/RApp/PICO-8%20(PICO-8%20standalone)/Roms/PICO/~Run%20PICO-8%20with%20Splore.png

XK9274, 2024-02-26:
Reproduced on one of mine after updating, removed the `PICO_cache6.db` from `/mnt/SDCARD/Roms/PICO/` and Refreshed roms with the menu item... Seems to work fine after this - can you confirm?

If not can you send me your:

`/mnt/SDCARD/RApp/PICO-8/launch.sh` 
file, and
`/mnt/SDCARD/Roms/PICO/` 
directory archived as .7z or .zip (remove any carts that may be either paid or licensed please)

XK9274, 2024-05-08:
Are you still having this issue?

closed 2024-06-23 by Aemiii91

TotallyNotSethP, 2024-07-09:
Sorry had a really stressful move and forgot about this but removing PICO_cache6.db did the trick! Thanks to everyone who helped me out :)