OnionUI/Onion #1458: no audio initialization on Miyoo Mini V2

md0-code, 2024-02-29:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.3.0

### Miyoo Firmware Version

202306111426

### Provide a clear and concise description of the issue

Upon starting/restarting the device there is no sound in the meniu or from most of the Retroarch cores until specific cores or apps run for the first time.

### Steps to Reproduce

Some cores (Scummvm, for example) or apps (Duke Nukem 3D) can activate the sound, in two steps - on first launch first there is an audible click (initialization noise) at first but still no sound, but on subsequent launches sound is working everywhere.

### Provide any additional context or information that may help us investigate your issue

Tested on Miyoo Mini V2

Aemiii91, 2024-02-29:
Can you attach a video showcasing the issue?

md0-code, 2024-02-29:
I think I've found the problem. After moving the SD card to a Miyoo Mini+ I saw that the sound was set to 0 (muted). Raising the volume on the Mini+ and moving the SD card back to the Mini apparently fixed the issue. Could it be that the Minis' physical rocker does not clear the mute when pushed from the lower position?

Aemiii91, 2024-02-29:
Yes, definitely. We might need to disable mute on MM, though I thought we already did 🤷‍♂️

tGecko, 2024-03-01:
We indeed have that in runtime [here](https://github.com/OnionUI/Onion/blob/b5b77c2958dbad6a62e71e33082e9414d9993cad/static/build/.tmp_update/runtime.sh#L733) - wonder why it doesn't seem to work: