OnionUI/Onion #1491: Dropbear not accepting connecton

alexzeitgeist, 2024-03-09:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.3.0

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

When attempting to establish an SSH connection, the process fails unexpectedly. The connection is terminated by the server immediately after the SSH2_MSG_KEXINIT exchange, without proceeding to the expected SSH2_MSG_KEX_ECDH_REPLY phase. This issue persists regardless of whether SSH authentication is enabled. Initial troubleshooting steps, including adjustments to the MTU settings on the client side, have not resolved the problem. Further investigation revealed a potential issue related to the creation of temporary host keys by Dropbear.

### Steps to Reproduce

1. Enable SSH in OnionUI, testing both with and without authentication.
2. From the client side, attempt to connect to the server:
`ssh -vvvvvv -oKexAlgorithms=diffie-hellman-group14-sha1 -oHostKeyAlgorithms=ssh-rsa,ssh-dss -p 22 root@192.168.120.43 -m hmac-sha1`
3. Observe that the connection is closed by the server right after the key exchange initialization, without proceeding to the key exchange reply phase:
```
OpenSSH_8.9p1 Ubuntu-3ubuntu0.6, OpenSSL 3.0.2 15 Mar 2022
debug1: Reading configuration data /home/alex/.ssh/config
debug1: /home/alex/.ssh/config line 580: Applying options for *
debug3: kex names ok: [curve25519-sha256@libssh.org,diffie-hellman-group-exchange-sha256]
debug1: Reading configuration data /etc/ssh/ssh_config
debug3: /etc/ssh/ssh_config line 19: Including file /etc/ssh/ssh_config.d/authentication.conf depth 0
debug1: Reading configuration data /etc/ssh/ssh_config.d/authentication.conf
debug3: /etc/ssh/ssh_config line 19: Including file /etc/ssh/ssh_config.d/ciphers.conf depth 0
debug1: Reading configuration data /etc/ssh/ssh_config.d/ciphers.conf
debug3: kex names ok: [curve25519-sha256@libssh.org,ecdh-sha2-nistp521,ecdh-sha2-nistp384,ecdh-sha2-nistp256,diffie-hellman-group-exchange-sha256]
debug3: /etc/ssh/ssh_config line 19: Including file /etc/ssh/ssh_config.d/options.conf depth 0
debug1: Reading configuration data /etc/ssh/ssh_config.d/options.conf
debug3: /etc/ssh/ssh_config line 19: Including file /etc/ssh/ssh_config.d/reenable_rsa.conf depth 0
debug1: Reading configuration data /etc/ssh/ssh_config.d/reenable_rsa.conf
debug1: /etc/ssh/ssh_config line 21: Applying options for *
debug2: resolve_canonicalize: hostname 192.168.120.43 is address
debug3: expanded UserKnownHostsFile '~/.ssh/known_hosts' -> '/home/alex/.ssh/known_hosts'
debug3: expanded UserKnownHostsFile '~/.ssh/known_hosts2' -> '/home/alex/.ssh/known_hosts2'
debug3: ssh_connect_direct: entering
debug1: Connecting to 192.168.120.43 [192.168.120.43] port 22.
debug3: set_sock_tos: set socket 3 IP_TOS 0x10
debug1: Connection established.
debug1: identity file /home/alex/.ssh/id_rsa type 0
debug1: identity file /home/alex/.ssh/id_rsa-cert type -1
debug1: identity file /home/alex/.ssh/id_ecdsa type -1
debug1: identity file /home/alex/.ssh/id_ecdsa-cert type -1
debug1: identity file /home/alex/.ssh/id_ecdsa_sk type -1
debug1: identity file /home/alex/.ssh/id_ecdsa_sk-cert type -1
debug1: identity file /home/alex/.ssh/id_ed25519 type 3
debug1: identity file /home/alex/.ssh/id_ed25519-cert type -1
debug1: identity file /home/alex/.ssh/id_ed25519_sk type -1
debug1: identity file /home/alex/.ssh/id_ed25519_sk-cert type -1
debug1: identity file /home/alex/.ssh/id_xmss type -1
debug1: identity file /home/alex/.ssh/id_xmss-cert type -1
debug1: identity file /home/alex/.ssh/id_dsa type -1
debug1: identity file /home/alex/.ssh/id_dsa-cert type -1
debug1: Local version string SSH-2.0-OpenSSH_8.9p1 Ubuntu-3ubuntu0.6
debug1: Remote protocol version 2.0, remote software version dropbear_2022.83-MM
debug1: compat_banner: no match: dropbear_2022.83-MM
debug2: fd 3 setting O_NONBLOCK
debug1: Authenticating to 192.168.120.43:22 as 'root'
debug3: send packet: type 20
debug1: SSH2_MSG_KEXINIT sent
debug3: receive packet: type 20
debug1: SSH2_MSG_KEXINIT received
debug2: local client KEXINIT proposal
debug2: KEX algorithms: diffie-hellman-group14-sha1,ext-info-c,kex-strict-c-v00@openssh.com
debug2: host key algorithms: ssh-rsa,ssh-dss
debug2: ciphers ctos: chacha20-poly1305@openssh.com,aes256-gcm@openssh.com,aes128-gcm@openssh.com,aes256-ctr,aes192-ctr,aes128-ctr
debug2: ciphers stoc: chacha20-poly1305@openssh.com,aes256-gcm@openssh.com,aes128-gcm@openssh.com,aes256-ctr,aes192-ctr,aes128-ctr
debug2: MACs ctos: hmac-sha1
debug2: MACs stoc: hmac-sha1
debug2: compression ctos: none,zlib@openssh.com,zlib
debug2: compression stoc: none,zlib@openssh.com,zlib
debug2: languages ctos:
debug2: languages stoc:
debug2: first_kex_follows 0
debug2: reserved 0
debug2: peer server KEXINIT proposal
debug2: KEX algorithms: curve25519-sha256,curve25519-sha256@libssh.org,ecdh-sha2-nistp521,ecdh-sha2-nistp384,ecdh-sha2-nistp256,diffie-hellman-group14-sha1,diffie-hellman-group1-sha1,kexguess2@matt.ucc.asn.au
debug2: host key algorithms: ssh-ed25519,ecdsa-sha2-nistp256,rsa-sha2-256,ssh-rsa,ssh-dss
debug2: ciphers ctos: aes128-ctr,aes128-cbc,3des-ctr,3des-cbc
debug2: ciphers stoc: aes128-ctr,aes128-cbc,3des-ctr,3des-cbc
debug2: MACs ctos: hmac-sha1
debug2: MACs stoc: hmac-sha1
debug2: compression ctos: none
debug2: compression stoc: none
debug2: languages ctos:
debug2: languages stoc:
debug2: first_kex_follows 0
debug2: reserved 0
debug1: kex: algorithm: diffie-hellman-group14-sha1
debug1: kex: host key algorithm: ssh-rsa
debug1: kex: server->client cipher: aes128-ctr MAC: hmac-sha1 compression: none
debug1: kex: client->server cipher: aes128-ctr MAC: hmac-sha1 compression: none
debug2: bits set: 1021/2048
debug3: send packet: type 30
debug1: expecting SSH2_MSG_KEX_ECDH_REPLY
Connection closed by 192.168.120.43 port 22
```

5. Modify the MTU settings on the client side and attempt to reconnect, observing that the issue persists.

### Provide any additional context or information that may help us investigate your issue

I compiled and run a local instance of dropbear-2022.83 with logging enabled (dropbear -R -F) and attempted to connect again. Same issue as with dropbear in OnionUI. The debug log from indicates a failure to create a new temporary file for the host key due to the absence of the /etc/dropbear directory. This results in the connection being terminated before authentication:

```
Mar 09 23:27:46 thinkpad-linux dropbear[258102]: Couldn't create new file /etc/dropbear/dropbear_rsa_host_key.tmp258102: No such file or directory
Mar 09 23:27:46 thinkpad-linux dropbear[258102]: Exit before auth from <192.168.120.69:52208>: Couldn't read or generate hostkey /etc/dropbear/dropbear_rsa_host_key
```

Creating the missing /etc/dropbear directory and restarting Dropbear resolves the issue, allowing for successful SSH login. This suggests that a similar directory creation or permissions issue might be present with Dropbear within the OnionUI, preventing the successful establishment of SSH connections.

tGecko, 2024-03-10:
Hi, thanks for the very detailed bug report.

Have you tried connecting without specifying any extra params?

Just using `ssh root@192.168.120.43` should work.

The key path is moved to `/mnt/SDCARD/.tmp_update/etc/dropbear/` because we don't have write access on /etc/dropbear, which lives on spi flash.

We use a [modified version](https://github.com/tGecko/dropbear) of dropbear 2022.83 with some changes made to fit the MMP:

### Changes made:
* based on [dropbear 2022.83](https://github.com/mkj/dropbear/tree/DROPBEAR_2022.83)
* every user logs in as root
* shell is started as interactive shell to avoid loading of /etc/profile
* for blank password, check the actual submitted password length instead of checking the password hash length
* don't chdir into home directory
* change LD_LIBRARY_PATH
* move ssh keys to /mnt/SDCARD/.tmp_update/etc/dropbear/

nevertheless I will have a look at why connecting with your specified arguments fails.

tGecko, 2024-03-10:
Indeed I have no trouble connecting with your parameters
```
thomas@PC:~/code/dropbear$ ssh -oKexAlgorithms=diffie-hellman-group14-sha1 -oHostKeyAlgorithms=ssh-rsa,ssh-dss -p 22 root@purple -m h
mac-sha1
The authenticity of host 'purple (10.0.0.28)' can't be established.
RSA key fingerprint is SHA256:8v9Mjm8U2937rRGJvW+IL6j5V7p2coP9Gk77lPw45/w.
This key is not known by any other names
Are you sure you want to continue connecting (yes/no/[fingerprint])? yes
Warning: Permanently added 'purple' (RSA) to the list of known hosts.
root@purple's password:
/ #```

alexzeitgeist, 2024-03-10:
> The key path is moved to `/mnt/SDCARD/.tmp_update/etc/dropbear/` because we don't have write access on /etc/, which lives on spi flash. Does this directory exist on your device?

Hah! I didn't have /.tmp_update/etc/dropbear/, not even /.tmp_update/etc/. Manually creating the empty directory did the trick, thanks. :) The reason: I use git to keep track of changes on my sdcard; obviously, since the path is just an empty path, it is not added to the repo. Stupid me.

May I suggest adding an empty .gitkeep file inside /mnt/SDCARD/.tmp_update/etc/dropbear/ as part of the initial setup routine? 

onion.pak:
```
$ find . -type d -empty
./Screenshots
./Roms
./App/PackageManager/data/RApp/Uzebox (Uzem)/Roms/UZEBOX
./App/PackageManager/data/RApp/Thomson - MOTO (Theodore)/Roms/THOMSON
./App/PackageManager/data/RApp/Sony - PlayStation (PCSX standalone)/Roms/PS
./App/PackageManager/data/RApp/Sinclair - ZX81 (81)/Roms/ZXEIGHTYONE
./App/PackageManager/data/RApp/Sharp - X68000 (PX68k)/Roms/X68000
./App/PackageManager/data/RApp/Sega - Visual Memory Unit (VeMUlator)/Roms/VMU
./App/PackageManager/data/RApp/Sega - Sega CD (Genesis Plus GX)/Roms/SEGACD
./App/PackageManager/data/RApp/Sega - Roms Hacks (Genesis Plus GX)/Roms/MDHACKS
./App/PackageManager/data/RApp/Sega - Master System (Genesis Plus GX)/Roms/MS
./App/PackageManager/data/RApp/Sega - Master System (Gearsystem)/Roms/MS
./App/PackageManager/data/RApp/Sega - Genesis (Genesis Plus GX)/Roms/MD
./App/PackageManager/data/RApp/Sega - Game Gear (Genesis Plus GX)/Roms/GG
./App/PackageManager/data/RApp/Sega - Game Gear (Gearsystem)/Roms/GG
./App/PackageManager/data/RApp/Palm OS - m515 (mu)/Roms/PALM
./App/PackageManager/data/RApp/PICO-8 (Fake8 standalone)/Roms/PICO
./App/PackageManager/data/RApp/Nintendo - SNES (Snes9x)/Roms/SFC
./App/PackageManager/data/RApp/Nintendo - SNES (Snes9x 2010)/Roms/SFC
./App/PackageManager/data/RApp/Nintendo - SNES (Snes9x 2005)/Roms/SFC
./App/PackageManager/data/RApp/Nintendo - SNES (Snes9x 2005 Plus)/Roms/SFC
./App/PackageManager/data/RApp/Nintendo - SNES (Snes9x 2002)/Roms/SFC
./App/PackageManager/data/RApp/Nintendo - SNES (ChimeraSNES)/Roms/SFC
./App/PackageManager/data/RApp/Nintendo - NES (Nestopia)/Roms/FC
./App/PackageManager/data/RApp/Nintendo - GBC (TGB Dual)/Roms/GBC
./App/PackageManager/data/RApp/Nintendo - GBC (Gearboy)/Roms/GBC
./App/PackageManager/data/RApp/Nintendo - GBA (gpSP)/Roms/GBA
./App/PackageManager/data/RApp/Nintendo - GBA (VBA-M)/Roms/GBA
./App/PackageManager/data/RApp/Nintendo - GBA (VBA Next)/Roms/GBA
./App/PackageManager/data/RApp/Nintendo - GBA (Meteor)/Roms/GBA
./App/PackageManager/data/RApp/Nintendo - GBA (Mednafen GBA)/Roms/GBA
./App/PackageManager/data/RApp/Nintendo - GB (TGB Dual)/Roms/GB
./App/PackageManager/data/RApp/Nintendo - GB (Gearboy)/Roms/GB
./App/PackageManager/data/RApp/NEC - PC-FX (Mednafen PC-FX)/Roms/PCFX
./App/PackageManager/data/RApp/NEC - PC-98 (Neko Project II)/Roms/PCNINETYEIGHT
./App/PackageManager/data/RApp/NEC - PC-98 (Neko Project II Kai)/Roms/PCNINETYEIGHT
./App/PackageManager/data/RApp/NEC - PC-8000 (Quasi88)/Roms/PCEIGHTYEIGHT
./App/PackageManager/data/RApp/Microsoft - MSX (fMSX)/Roms/MSX
./App/PackageManager/data/RApp/Microsoft - DOS (DOSBox-Pure 0.9.7)/Roms/DOS
./App/PackageManager/data/RApp/Game engine - Open Beats of Rage/Roms/OPENBOR
./App/PackageManager/data/RApp/Game engine - MicroW8 (Uw8)/Roms/MICROW8
./App/PackageManager/data/RApp/Game engine - LÖVE (Lutro)/Roms/LUTRO
./App/PackageManager/data/RApp/Game engine - EasyRPG (EasyRPG)/Roms/EASYRPG
./App/PackageManager/data/RApp/Game engine - ChaiLove (ChaiLove)/Roms/CHAI
./App/PackageManager/data/RApp/Game Music Emu (GME)/Roms/GME
./App/PackageManager/data/RApp/Commodore - Sharp X1 (x1)/Roms/XONE
./App/PackageManager/data/RApp/Commodore - Amiga CD32 (uae4arm)/Roms/AMIGACD
./App/PackageManager/data/RApp/Commodore - Amiga (uae4arm)/Roms/AMIGA
./App/PackageManager/data/RApp/BK - 0010 (BK)/Roms/EBK
./App/PackageManager/data/RApp/Atari - ST (hatari)/Roms/ATARIST
./App/PackageManager/data/RApp/Atari - Lynx (Mednafen Lynx)/Roms/LYNX
./App/PackageManager/data/RApp/Atari - Jaguar (Virtual Jaguar)/Roms/JAGUAR
./App/PackageManager/data/RApp/Atari - 800 (atari800)/Roms/EIGHTHUNDRED
./App/PackageManager/data/RApp/Atari - 5200 (atari800)/Roms/FIFTYTWOHUNDRED
./App/PackageManager/data/RApp/Arduboy (Arduous)/Roms/ARDUBOY
./App/PackageManager/data/RApp/Arcade - LaserDisk (Daphne)/Roms/DAPHNE
./App/PackageManager/data/RApp/Arcade (MAME 2003-xtreme)/Roms/ARCADE
./App/PackageManager/data/RApp/Arcade (MAME 2000)/Roms/MAME2000
./App/PackageManager/data/RApp/Arcade (M.B.A-mini)/Roms/MBA
./App/PackageManager/data/RApp/Arcade (Final Burn Neo)/Roms/FBNEO
./App/PackageManager/data/RApp/Arcade (FB Alpha 2012)/Roms/FBA2012
[... 41 lines cut ...]
./App/PackageManager/data/Emu/Fairchild - ChannelF (FreeChaF)/Roms/FAIRCHILD
./App/PackageManager/data/Emu/Commodore - VIC-20 (VICE xvic)/Roms/VIC20
./App/PackageManager/data/Emu/Commodore - C64 (VICE x64)/Roms/COMMODORE
./App/PackageManager/data/Emu/Commodore - Amiga (PUAE 2021)/Roms/AMIGA
./App/PackageManager/data/Emu/Coleco - ColecoVision (blueMSX)/Roms/COLECO
./App/PackageManager/data/Emu/Capcom - CPS3 (FB Alpha 2012 CPS-3)/Roms/CPS3
./App/PackageManager/data/Emu/Capcom - CPS2 (FB Alpha 2012 CPS-2)/Roms/CPS2
./App/PackageManager/data/Emu/Capcom - CPS1 (FB Alpha 2012 CPS-1)/Roms/CPS1
./App/PackageManager/data/Emu/Bandai - WonderSwanColor (B. Cygne)/Roms/WS
[... cut ...]
./App/PackageManager/data/Emu/Atari - 2600 (Stella 2014)/Roms/ATARI
./App/PackageManager/data/Emu/Arcade (MAME 2003-Plus)/Roms/ARCADE
./App/PackageManager/data/Emu/Amstrad - CPC (CrocoDS)/Roms/CPC
./App/PackageManager/data/App/PDF Reader (Green)/Media/PDF
./.tmp_update/etc/dropbear
```

I don't think the empty /App/PackageManager/data/Emu/ paths are required, at least I never had issues even when they were not present. So it's really just /.tmp_update/etc/dropbear that would fail on me.

Thanks again and sorry for the trouble.

Alex

PR #1493 (2024-04-12): Fix SSH not connecting if key directory missing
If the directory `/mnt/SDCARD/.tmp_update/etc/dropbear` is missing, connecting via SSH fails because dropbear can't create it's host keys. Creates the directory if needed.

Closes #1491
commit 3ad5f32b (2024-04-12): Fix SSH not connecting if key directory missing (#1493)

If the directory `/mnt/SDCARD/.tmp_update/etc/dropbear` is missing,
connecting via SSH fails because dropbear can't create it's host keys.
Creates the directory if needed.

Closes #1491
closed 2024-04-12 by tGecko