OnionUI/Onion #1516: Delayed application of overlays when wifi enabled

bmn001, 2024-03-20:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.3.0

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

I've been using overlays for a few systems.  For this example, I've been using this great one for Game Boy: https://www.reddit.com/r/MiyooMini/comments/18e2o0z/i_remastered_my_game_boy_dmg_overlay/

When I launch a game with wifi disabled, the overlay is immediately visible and active as expected.  No problem.

When I launch a game with wifi enabled and a retroachievements login configured, the overlay is *sometimes* immediately active.  However sometimes it won't turn itself on automatically for several _seconds_.  In other words, the game will run without _any_ overlay for a while, and then it'll just turn itself on a little while later.

A few seconds (<10) seems to be the norm, but once it took almost a full minute.

### Steps to Reproduce

- Enable the overlay listed above using all steps provided in the reddit post.
- Ensure your RetroAchievements login information is stored in Retroarch and you're within range of a stored wireless network.
- Launch any Game Boy game.

You may not experience it right away.  It may take a few launches in various wifi environments to occur.

### Provide any additional context or information that may help us investigate your issue

It's almost like the system first runs the rom, then tries to confirm that it has wifi and/or a connection to the RetroAchievements servers (or times out), and only _then_ will it apply the overlay.

I don't know if this is a general wifi stack issue or an internal Retroarch issue.  This has never happened to me on any other hardware platform when using Retroarch (Raspberry Pi, Ubuntu, Windows) so figured it was worth a bug report here.

bmn001, 2024-03-25:
After extensive testing I can say that this behavior only occurs when Retroarch has achievements enabled.

**Disabling achievements** causes overlays to appear correctly, immediately, every single time.

I've updated the title of the bug to reflect this.

XK9274, 2024-03-25:
Thanks, have you checked if this bug is present in the stock version of RA that miyoo ships, or whether similar bug reports exist on the RA repo?

bmn001, 2024-03-26:
I went back to the stock Miyoo OS to give that a test, and the version of Retroarch they included doesn't include support for Achievements at all.  Achievements were an Onion addition.

I spent some time searching through the Retroarch issue list for any similar bug reports (including closed bugs) and saw none.  Doesn't mean I missed one though - Retroarch currently has 2593 open issues!  :woozy_face:  So far the MM+ is the only Retroarch platform where I've experienced this achievement/overlay delay.

pcone, 2025-11-06:
I'm seeing this too on the latest beta FW. It's extra annoying when you don't always have wifi because the overlay only shows once RA finally gives up trying to connect which takes a while.

RdrSeraphim, 2026-01-18:
I haven't been able to replicate this and I use both core and content directory overrides.

RetroAchievements in some cases has a ton of data to pull when a game starts which can cause quite a freeze at the beginning. Pokemon Emerald was recently overhauled to have just shy of 200 achievements. I wouldn't be surprised if, for some, overlays aren't being applied simply because RetroAchievements locks the entire emulator until the achievements are pulled.