OnionUI/Onion #1518: Loading issue on directory of a game

Alessandroinfo, 2024-03-20:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.3

### Miyoo Firmware Version

202305052130

### Provide a clear and concise description of the issue

When i open some folder game in Onion OS v 4.3 i see a list of "Loading" instead of the game file. Someone know how to fix this?
[image]

### Steps to Reproduce

Go to a game folder that contains a game (PS1)

### Provide any additional context or information that may help us investigate your issue

I installed Onion from scratch. After updated with 4.3 version. The issue was also on the previous version (4.2.3)

LinkSlayer64, 2024-03-22:
Chiming in to say this is happening to me, and I think I narrowed it down. All of my games are in individual folders due to being mirrored from my PC's no-intro or redump sets.
In each instance, the folder that loads into a list of "loading" items has an Apostrophe `'` in it. I.E. "Kirby's Adventure" or "Spyro 2 - Ripto's Rage"

LinkSlayer64, 2024-03-22:
Adding an update, this is apparently a miyoo OS issue, referenced here : #301 
It is listed in the docs here: https://onionui.github.io/docs/emulators#rom-files
Although maybe a description of what using an apostrophe does for those searching the symptom, and/or a "known-issues" page or something might be more eyecatching, but you all do you.
You'll have to forgive me, I was working backwards from "loading" to realizing the issue.

Alessandroinfo, 2024-03-23:
> Adding an update, this is apparently a miyoo OS issue, referenced here : #301 It is listed in the docs here: https://onionui.github.io/docs/emulators#rom-files Although maybe a description of what [...]

Yes this is the issue. With which character i can replace the '?

Phanturian, 2024-03-24:
I would recommend one of two things. If you're using .bin and .cue files, get rid of the individual folders and put all the .bin and .cue files directly into the Roms/PSX folder. The other option is to use PS1 games in the .chd format, where the .bin and .cue files have been compressed into a single lossless file. I was using .bin and .cue, but now I only use .chd because the file size is much smaller, and there are less files. You can also convert your .bin and .cue files to .chd using CHDMAN.

Alessandroinfo, 2024-03-24:
> I would recommend one of two things. If you're using .bin and .cue files, get rid of the individual folders and put all the .bin and .cue files directly into the Roms/PSX folder. The other option is [...]

Many thanks about your advice. Also i replace the apostrophe with a dash while waiting for a fix.