OnionUI/Onion #1519: ScummVM Standalone doesn't support all CD audio formats

bmn001, 2024-03-21:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.3.0

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

**TLDR:** .flac and .m4a audio doesn't currently work in standalone ScummVM.

Per the [ScummVM documentation](https://docs.scummvm.org/en/v2.7.1/use_scummvm/game_files.html#cd-audio), any games that shipped with CD audio tracks need to have those tracks ripped into one of four supported file formats: .mp3, .flac, .m4a, or .ogg.

The current standalone version of ScummVM included with Onion (2.9.0git) seems to only support _two_ of these file formats: **.ogg** and **.mp3**.  When audio is encoded in either format, games work fine.

However, if you attempt to launch a game with **.flac** audio tracks, the game will launch but you'll see this error: "This game has audio tracks on its CD.  These tracks need to be ripped from the CD using an appropriate CD audio extracting tool in order to listen to the game's music.  See the documentation (CD audio) for details."  The game will be playable, but with no CD audio.

If you attempt to launch a game with **.m4a** audio tracks, the game will launch and you _won't_ receive the error, but it will also not contain CD audio.

### Steps to Reproduce

- Load your ScummVM roms directory with "The Secret of Monkey Island (CD DOS VGA)" version as a test.  Or try any other game containing ripped CD audio tracks.
- Ensure the audio tracks are encoded as .flac or .m4a.  If they are not currently in these formats, convert the audio tracks so that they are.
- Launch the game.

With .flac, you'll get an error message and no CD audio playback.  With .m4a you won't get an error message, but you also won't have CD audio playback.

### Provide any additional context or information that may help us investigate your issue

The ScummVM libretro core (2.7.0git) works as expected and supports all four file formats.  This bug report only impacts the standalone/expert version.

XK9274, 2024-04-09:
Hi, 

Can you try this build please: https://www.dropbox.com/scl/fi/yhuqsqnm07ckujzti3tjm/scummvm.zip?rlkey=76hhg1o8720r2gjbau15y4sd4&dl=0

## **I've added:**
FLAC
FAAD
A52

I see no options for m4a, though FAAD should do it no? If there's other options that need enabling please do comment and we'll see what we can do about expanding ScummVM 

Just~ FYI - GIF is compiled but does not get picked up, i'll check this another day.

And compiled with `--enable-mt32emu` 

This is a synced fork of scummvm so should be up to date with the master branch. If you find bugs please let us know - If not i'll move to PR. 

As for your other issue here: https://github.com/OnionUI/Onion/issues/1510 - I'm not getting into the dependency hole that is FluidSynth and Glib2 atm

## Other

@Leemann07 - Please can you check if the missing commits are in this build from our previous conversations?

## Compile log

<details>

```
Running ScummVM configure...
Looking for C++ compiler... arm-linux-gnueabihf-g++
Checking if compiler supports C++11...
yes
Checking for compiler version... GCC 8.3.0, ok
Checking if C++11 initializer list is available... yes
Checking if C++11 std::nullptr_t is available...
yes
Checking if C++11 alignas and alignof keywords are available... yes
Using LLD linker... no
Using Mold linker... no
Checking for whether C++ compiler accepts -Wglobal-constructors... no
Checking for whether C++ compiler accepts -Wno-undefined-var-template... no
Checking for whether C++ compiler accepts -Wno-pragma-pack... no
Checking for whether C++ compiler accepts -Wno-address-of-packed-member... no
Checking endianness... little
Checking 64-bitness... no
Type with 1 byte... char
Type with 2 bytes... short
Type with 4 bytes... int
Type with 8 bytes... long long
Alignment required... yes
Checking host CPU architecture... ARM
Checking hosttype... linux
Cross-compiling to miyoomini
Checking for pkg-config... yes
WARNING: When cross-compiling PKG_CONFIG_LIBDIR must be set to the location of the .pc files for the target
Looking for sdl-config... /opt/miyoomini-toolchain/bin/../arm-linux-gnueabihf/libc/usr/bin/sdl-config
Checking for SDL... yes
Checking for SDL_Net... no
Checks for ENet...
   Checking for sendmsg... yes
   Checking for recvmsg... yes
Optional checks for ENet...
   Checking for getaddrinfo... yes
   Checking for getnameinfo... yes
   Checking for gethostbyaddr_r... yes
   Checking for gethostbyname_r... yes
   Checking for poll... yes
   Checking for fcntl... yes
   Checking for inet_pton... yes
   Checking for inet_ntop... yes
[... 280 lines cut ...]
  adding: App/scummvm/scummvm.png (deflated 0%)
  adding: App/scummvm/config.json (deflated 24%)
  adding: App/scummvm/scummvm (deflated 59%)
  adding: App/scummvm/launch.sh (deflated 31%)
  adding: App/scummvm/vkeybd_default.zip (stored 0%)
  adding: App/scummvm/COPYING.GLAD (deflated 62%)
  adding: App/scummvm/COPYING.MIT (deflated 63%)
  adding: App/scummvm/COPYING.MKV (deflated 48%)
  adding: App/scummvm/COPYING.FREEFONT (deflated 66%)
  adding: App/scummvm/themes/ (stored 0%)
  adding: App/scummvm/themes/shaders.dat (deflated 11%)
  adding: App/scummvm/themes/scummclassic.zip (stored 0%)
  adding: App/scummvm/themes/scummremastered.zip (stored 0%)
  adding: App/scummvm/themes/residualvm.zip (stored 0%)
  adding: App/scummvm/themes/scummmodern.zip (stored 0%)
  adding: App/scummvm/themes/gui-icons.dat (deflated 0%)
  adding: App/scummvm/themes/translations.dat (deflated 65%)
  adding: App/scummvm/COPYING.TINYGL (deflated 42%)
  adding: App/scummvm/CONTRIBUTING.md (deflated 47%)
  adding: App/scummvm/COPYING (deflated 66%)
  adding: App/scummvm/COPYING.BSD (deflated 74%)
root@cc5d71992d6f:~/workspace/scummvm-miyoo#
```

</details>

XK9274, 2024-04-09:
Alright the new link is above, sorry about that - had to recompiled.

bmn001, 2024-04-12:
Thanks so much for taking the time to make this new build.

I just tested with all four audio formats, and all four are now working.  flac, ogg, m4a, and mp3 all play CD audio with no error messages.

re: `--enable-mt32`, it didn't appear to change anything regarding #1510.  But that's a separate issue.

As far as _this_ issue is concerned, I'd consider it done and closeable.  Thank you!

XK9274, 2024-04-12:
> Thanks so much for taking the time to make this new build.
>
> I just tested with all four audio formats, and all four are now working. flac, ogg, m4a, and mp3 all play CD audio with no error [...]

More than happy to recompile, though i've had a massive data loss just after this so my ability may now be restricted until i get my build scripts back in order. Yeah fluidsynth is a bit of a pain and i'm not sure i'll get it working anytime soon .

When you've ran with this for a week or two can you let me know if you have any issues please and if not i'll move to PR it for 4.4

XK9274, 2024-05-08:
@bmn001 Has there been any issues? I'll soon recompile with zlib support as I forgot in the previous build so no zip files are accessible.

 If you don't mind testing this one too? Then we'll PR

bmn001, 2024-05-08:
No issues that I could see. Then previous build you provided is still going strong.

Just a quick side note - the docs might need an update.  They currently don't claim any kind of zip support for scummvm, and indicate that you need to create separate folders for each game. (https://onionui.github.io/docs/emulators/scummvm)

I'm happy to test whatever you've got!

XK9274, 2024-05-08:
Yeah, pretty sure at one point I added zlib support, then at another removed the zlib configure/compile and I don't remember why, was maybe it wasn't quite fully functional.

I'll do another full compile with all our previous mods here aswell as zlib support so hopefully the zip files will work again & I'll include changes to docs in the PR, if it turns out to not work at all we'll just go ahead with the last working build. 

Thanks!

bmn001, 2024-05-30:
One week in on this new build and things seem be to running well!

There is only one display issue that I can see, and I think it might be more on the Onion side than the ScummVM side:

In some Scumm games (Secret of Monkey Island CD DOS is one, Police Quest II DOS is another), Onion's volume bar doesn't appear as a solid green line.  It's a green vertical line, but every other row of pixels in that line is black, and the whole bar flickers.

In these cases, a volume bar that shows "full" is actually only about 50% volume.  You can continue to press the volume up button and the volume will increase, but it won't be reflected in the on-screen bar.

bmn001, 2024-05-30:
A bit more info - this latest build fixes previously-unwatchable video playback in games like Myst, The 7th Guest, and Blade Runner!

I looked into zlib - ScummVM uses it to decompress .dxa videos in some games _and_ to compress/decompress save game data, but not to decompress the games themselves.  They've said  [they will never support zipped games](https://docs.scummvm.org/en/latest/help/faq.html#when-will-you-add-support-for-zip-archives) due to anti-piracy concerns.

breacher18, 2024-08-29:
I’m still having issues with broken sword 2.5 audio. Then in the stand alone audio is fine but cursor movement is so slow.