OnionUI/Onion #1520: Update EasyRPG core extension support

bmn001, 2024-03-21:
### Checklist

- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and something is incorrect or missing

### Wiki Section

https://onionui.github.io/docs/emulators/easyrpg

### Description

Current article says only .easyrpg extensions are supported.

They are, but the [official docs](https://docs.libretro.com/library/easyrpg/) for this core say that it also supports .ldb and .zip.

I've confirmed .zip files do work, just as long as the `RPG_RT.ini` and `RPG_RT.ldb` are in the root folder of the archive.

### Correction

Extensions: `.zip`, `.ldb`, `.easyrpg`

tGecko, 2024-03-21:
I think #1507 is what you're asking?

bmn001, 2024-03-21:
Ah yep, looks like!  Thank you!

Aemiii91, 2024-03-21:
Issue will be closed when PR is merged

closed 2024-05-09 by Aemiii91