OnionUI/Onion #1521: Documentation update request form is broken

bmn001, 2024-03-21:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

n/a

### Miyoo Firmware Version

n/a

### Provide a clear and concise description of the issue

When you create a new issue for the OnionUI project, you have the option of submitting either a bug report or a "Wiki Documentation Update Request" at https://github.com/OnionUI/Onion/issues/new/choose

When you fill out and submit the Documentation Update Request form, you're given a 404.  This is probably because the Github Wiki has been moved to https://onionui.github.io/docs/

### Steps to Reproduce

Submit the form at https://github.com/OnionUI/Onion/issues/new?assignees=&labels=documentation&projects=&template=wiki-update-request-only-new.yml&title=Wiki+Update+Title

### Provide any additional context or information that may help us investigate your issue

Here's the documentation update I just attempted to submit:

https://onionui.github.io/docs/emulators/easyrpg

Current article says only .easyrpg extensions are supported.

They are, but the [official docs](https://docs.libretro.com/library/easyrpg/) for this core say that it also supports .ldb and .zip.

I've confirmed .zip files do work, just as long as the `RPG_RT.ini` and `RPG_RT.ldb` are in the root folder of the archive.

Article correction should be:

Extensions: `.zip`, `.ldb`, `.easyrpg`

bmn001, 2024-03-21:
Uh okay so it's not broken.

While it did give a 404 after submitting, it did indeed make a new bug - https://github.com/OnionUI/Onion/issues/1520

XK9274, 2024-05-08:
Dunno if the 404 was related. But you are correct the template needs updating with `https://onionui.github.io/docs/` 

@Aemiii91 I don't have rights to do this without PR'ing it. Can you mod?

Aemiii91, 2024-05-09:
Template updated ✅

closed 2024-05-09 by Aemiii91