OnionUI/Onion #1539: Mini v4 screen only shows 750x559 of the internal 752x560

AnalogMan151, 2024-04-04:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.3.0

### Miyoo Firmware Version

202310271401

### Provide a clear and concise description of the issue

While working on some native Mini v4 overlays I noticed that despite the internal resolution being rendered at 752x560, and the screen shots taken while running games are taken in 752x560, that on the actual screen you cannot see the two left most columns of pixels nor the bottom row of pixels. They appear to be off screen. I've tried this on two separate Miyoo Mini v4 and found the same cropping on both.

I've attached 3 overlays that demonstrate the issue. One overlay has a single pixel wide border around it, another a 2px border and the last a 3px wide border.

[V4-Border_Test.zip](https://github.com/OnionUI/Onion/files/14863431/V4-Border_Test.zip)

### Steps to Reproduce

1. Extract attached V4 test overlays into overlay folder. 
2. Load up any RetroArch game. 
3. Open RetroArch menu and apply the 1px border test overlay and return to game. You will see the border only on the top and right side.
4. Reopen RetroArch menu and apply the 2px border test overlay and return to game. You will see a thicker border on the top and right and a thinner border along the bottom.
5.  Reopen RetroArch menu and apply the 3px border test overlay and return to game. You will see the thickest border on the top and right, thick border along the bottom and a thin border along the left.

What you should be seeing is an equal border of increasing thickness on all four sides as you apply each successive overlay.

### Provide any additional context or information that may help us investigate your issue

It appears that although the V4 driver produces a 752x560 image the screen itself only has a view window of 750x559. Since the viewport and viewport offsets are set in the SDL graphics driver in the custom forked RetroArch repository we can't change the viewport settings ourselves. If this issue holds true for all V4 models and not just my two, then resX and resY will need to be adjusted to `resX -= 2` and `resY -= 1` along with `vid->video_x = 2`, `vid->video_y = 1` and `vinfo.xoffset = 2`, `vinfo.yoffset = 1` and probably some others.

eCorke, 2024-10-03:
The problem is also clearly visible in main menu when using theme Analogue where selection boxes are cut by 2 pixels on the left. My model is Miyoo mini V4 without sticker (I've seen a dispute about actual version and sometimes being called V4.1) and firmware 202310271401.

SaulSNES, 2025-04-01:
> The problem is also clearly visible in main menu when using theme Analogue where selection boxes are cut by 2 pixels on the left. My model is Miyoo mini V4 without sticker (I've seen a dispute about actual version and sometimes being called V4.1) and firmware 202310271401.

I returned one V4 because of this same issue with the same theme, bought another and find myself with the same issue. Were you ever able to figure this out?