OnionUI/Onion #1551: Pico-8 Standalone Black screen with Sound

davidoppen, 2024-04-12:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.3

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

After upgrading to 4.3 and installing pico-8 standalone, I managed to play several pico-8 games. After a while (I don't know if I made any changes) pico-8 games and splores only open to a blackscreen. 

When directly opening a pico-8 game from the splore folder (bas-8) the game launches with a black screen and audio. I can control the game and enter the menu (by pressing 'start') and use the shortcuts (https://onionui.github.io/docs/emulators/pico-8-standalone#shortcuts).

I already removed and reinstalled the pico-8 standalone with the package-manager.

### Steps to Reproduce

1. Enter Expert screen
2. Enter Pico-8 Standalone
3. Enter Pico-8 Folder 
4. Start any Game 
5. Black Screen with sound and working controlls

### Provide any additional context or information that may help us investigate your issue

I will provide a LOG if that helps (and someone tells me where to access the log-files in the file structure.

XK9274, 2024-04-12:
Heya,

Your config.txt file is probably broken.

https://github.com/OnionUI/Onion/blob/main/static%2Fconfigs%2FSaves%2FCurrentProfile%2Fsaves%2FPICO-8%2F.lexaloffle%2Fpico-8%2Fconfig.txt

Download or copy the contents of this file into:
Saves/CurrentProfile/saves/PICO-8/.lexaloffle/pico-8/

davidoppen, 2024-04-12:
@XK9274 What should I do if there is no PICO-8 folder inside the Saves/Current-Profile/saves folder?

XK9274, 2024-04-12:
Do you have hidden files and folders turned on, on whatever device you're using to open the SD card?

You should see a .lexaloffle directory

XK9274, 2024-04-12:
Have you tried creating the directory structure and dropping the config file in?

davidoppen, 2024-04-12:
Is this correct?

[image]

closed 2024-04-12 by davidoppen

XK9274, 2024-04-12:
> That did the trick! 
> 
> > Have you tried creating the directory structure and dropping the config file in?
> 
> 

Sorry fella at work, yeah that's it and great, glad it's working.

Just out of interest what lead up to this situation? Had you reinstalled/moved/deleted anything? Or had you moved from the wrapper version on my GH to the Onion version?

davidoppen, 2024-04-12:
[image]

@XK9274 Thanks for your help so far! This is what the screen looks now after using the config.txt you provided.

Is there a way to have it look like out of the box again? I think that the bezels are missing now.

XK9274, 2024-04-12:
The two versions are different and use different libraries. Ideally you should remove it with package manager and reinstall it, then place your Pico 8 licensed files into the bin directory 

# PICO-8

## PICO-8 standalone can now be installed from the Expert section in Package Manager. Remember to move your binary files into the new dirs:

- These binaries will need to be added to: /mnt/SDCARD/RApp/PICO-8/bin

- Any saves and config will go to: /mnt/SDCARD/Saves/CurrentProfile/saves/PICO-8

- Carts will follow the default Onion path: /mnt/SDCARD/Roms/PICO

- Splore carts will be stored in: /mnt/SDCARD/Roms/PICO/splore

- Favourites.txt (if you want to move it) should go in: /mnt/SDCARD/Saves/CurrentProfile/saves/PICO-8/.lexaloffle/pico-8/

davidoppen, 2024-04-12:
> Sorry fella at work, yeah that's it and great, glad it's working.
>
> Just out of interest what lead up to this situation? Had you reinstalled/moved/deleted anything? Or had you moved from the wrapper [...]

I'm sorry, I can't really replicate what I have done. I got into Pico-8 after I upgraded to Onion 4.3 and installing the 'Pico-8 standalone' from the package-manager. I remember that I removed and reinstalled 'Pico-8standalone' via the package-manager several times trying to fix the black screen issue, and installed 'fake-8' at one point. 

I'm not really versed in Onion-OS.

davidoppen, 2024-04-12:
"The two versions are different and use different libraries. Ideally you should remove it with package manager and reinstall it, then place your Pico 8 licensed files into the bin directory as it sounds like something has got a little confused."

Can you please elaborate? Is it enough to remove it, save and restart, then install it again with the package-manager (and add the pico-8 files for RasPi)? Do I need to remove the file structure I manually added up to this point? 

I'm sorry for not getting it fully : (

XK9274, 2024-04-12:
There's no need to apologise! it's quite alright. I was just wondering incase there was something we needed to fix. 

All the documentation for the standalone is here (including shortcuts for screen scale, bezels etc): https://onionui.github.io/docs/emulators/pico-8-standalone
Your Bezels and borders can be found here: `RApp/PICO-8/res`

If it's working you don't need to do anything else, just enjoy! 

Re: the libraries, the Onion files contain things specific to Onion, where-as the app wrapper was the original build before it got merged and they had slight differences but i've since started unifying them, thoguh that said the Onion version will always have different launch commands but you don't need to worry about any of this.

davidoppen, 2024-04-12:
After reinstalling Pico-8 standalone with the package-manager, the UI looks correct again. 

Thanks for the patience and help!

Have a nice day/night,

David from Germany

XK9274, 2024-04-12:
That's great! 

Take care David!