OnionUI/Onion #1552: HTTPS and SSH not working

adrienrichmond, 2024-04-12:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.3.0

### Miyoo Firmware Version

0628

### Provide a clear and concise description of the issue

When I select the HTTPS and SSH functionalities via the Tweaks app and return to the menu (to validate) this does not work. I also tried to ping my Miyoo and Windows puts me "time out".

### Steps to Reproduce

Please provide detailed steps that will help us replicate and investigate your issue : 

1. Enable the options 
2. Test this

### Provide any additional context or information that may help us investigate your issue

I am an operator at Free (Freebox delta) in Fiber.

XK9274, 2024-04-15:
PFSense?

johncpitcher, 2024-04-20:
Same issue. Via the terminal I have tried and am able to retrieve http://google.com but not https://google.com, the same on other sites - I get a certificate error.

closed 2024-06-23 by Aemiii91