OnionUI/Onion #1562: Checkoff scripts closed in 1 minute

Nerqs, 2024-04-22:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.3.0

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

I wrote a script in checkoff directory, but when i power off my miyoo mini plus this script is only running for 1 minute, after this the system force a power off.
I have tried my script in another dispositives and it works just fine.

### Steps to Reproduce

Make a script that keeps running more than a minute and put it in the checkoff directory.

### Provide any additional context or information that may help us investigate your issue

The shutdown function appears to be executed and scheduled before starting the scripts in the checkoff directory.

XK9274, 2024-05-08:
What's your script doing?

Nerqs, 2024-05-09:
> What's your script doing?

A backup script, but the system force a shutdown before this script finishes.

Nerqs, 2024-05-09:
No, the script just upload the Saves/CurrentProfile folder to the cloud.

XK9274, 2024-05-09:
How many saves do you have lol. 

How's your script setup? Ideally for all these files before moving them you should tar.gz the directory and move that up to the cloud instead as it'll be faster. 

Can't see otherwise how it would take over a minute to move files less than a meg each

Nerqs, 2024-05-09:
This is a good idea, but I've already resolved the file size issue. My question is whether there is a way to increase the time that the system forces shutdown or cancel this function, because if I try to include other scripts this function will prevent them from working.

Aemiii91, 2024-05-09:
Maybe the sleep timer is being triggered? You can try to get around this by setting the flag `/tmp/stay_awake`.

Nerqs, 2024-05-09:
Thank you very much for the answers, i made the changes and it worked perfectly.

closed 2024-05-09 by Nerqs

XK9274, 2024-05-09:
Thanks for letting us know, we've found what's causing it so it'll be fixed :)