OnionUI/Onion #1565: RetroArch resets Volume Gain (dB) when saving new overrides

dreams897, 2024-04-26:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.3.0 Beta

### Miyoo Firmware Version

202305052130

### Provide a clear and concise description of the issue

If you set RetroArch to have a different Volume Gain (dB) via Core overrides and save it, it will save correctly but if you load RetroArch again with the override and make additional changes then save the global overrides again the volume gain (dB) will be reset to zero in terms of output audio currently playing.

### Steps to Reproduce

Launch RetroArch --> Change Volume Gain (dB) to a different value and save Core overrides --> Load RetroArch again and make an additional change and save Core overrides again. While the settings will save it will also reset the current audio playing to 0 gain dB. 

### Provide any additional context or information that may help us investigate your issue

N/A

dreams897, 2024-04-26:
Sorry if this seems nitpicky, but it is a little annoying to have to launch RetroArch again to get it working correctly again then manually load some games that need to be properly launched via RetroArch.

Aemiii91, 2024-04-26:
Why does this not fall under the first item above?
> Any issues with the closed source frontend (MainUI), or RetroArch or any of the cores should be directed to their respective repositories

Phanturian, 2024-04-28:
If you want all your games to have the same Volume Gain (db), choose Save Core Overrides. If you want some games to have a different volume setting, choose Save Game Overrides. I just performed a test where I set the Volume Gain (db) to 2.0 and saved the Core Override. Then I picked two games where I set one volume to 4.0, the other to 6.0, and saved a Game Override for each. Those two specific games launch at their respective 4.0 and 6.0 volumes, and all the other games using the same core launch at 2.0 Volume Gain (db).

A Game Override will always take precedent over a Content Directory or Core Override, and should be used if you want different (db) values for different games. If you make a change to an existing Override, you MUST go back into the Overrides Menu and save it again before exiting the game.

So everything performs exactly as it should, as RetroArch was designed, using Onion 4.3.0. In no instances could I get it to reset to 0.0 (db) unless I chose Delete Game Overrides, Delete Core Overrides, or manually changed it back to 0.0 and re-saved the Override.

closed 2024-06-23 by Aemiii91

JimminyJim, 2024-10-15:
I have the same issue.