OnionUI/Onion #1568: NDS - DraStic german translation

Benman2785, 2024-04-29:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.3.0

### Miyoo Firmware Version

202303262339

### Provide a clear and concise description of the issue

i have a german translation for DraStic // https://github.com/steward-fu/nds
could you please add that?

### Steps to Reproduce

put it into /emu/nds/ressources/lang = its now german :)

### Provide any additional context or information that may help us investigate your issue

[german_deu.txt](https://github.com/OnionUI/Onion/files/15152255/german_deu.txt)

XK9274, 2024-05-08:
Hi, thanks for the translation.

Can you PR this please? Hopefully @tGecko can take a look as i believe he speaks German. 

If you fork the repo, add the file to the correct directory and then PR your changes we'll review.

Aemiii91, 2024-06-23:
Done

closed 2024-06-23 by Aemiii91