OnionUI/Onion #1580: Pico-8

LO-FANG, 2024-05-12:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.3

### Miyoo Firmware Version

20230628

### Provide a clear and concise description of the issue

Pico-8 splore could not connect to bbs

### Steps to Reproduce

Just copy my the bin file to my device and lunch splore 

### Provide any additional context or information that may help us investigate your issue

I try to test the url by curl this is what I got.
[res.txt](https://github.com/OnionUI/Onion/files/15285427/res.txt)

Aemiii91, 2024-05-12:
Important part of the output:
```sh
curl -v http://lexaloffle.com/bbs/cpost_lister3.php?cat=7
*   Trying 172.67.75.178:80...
* connect to 172.67.75.178 port 80 failed: Connection timed out
*   Trying 104.26.9.193:80...
* Connected to lexaloffle.com (104.26.9.193) port 80 (#0)
> GET /bbs/cpost_lister3.php?cat=7 HTTP/1.1

> Host: lexaloffle.com

> User-Agent: curl/8.1.0-DEV

> Accept: */*

> 

< HTTP/1.1 301 Moved Permanently

< Date: Sun, 12 May 2024 07:08:52 GMT

< Content-Type: text/html

< Content-Length: 167

< Connection: keep-alive

< Cache-Control: max-age=3600

< Expires: Sun, 12 May 2024 08:08:52 GMT

< Location: https://www.lexaloffle.com/bbs/cpost_lister3.php?cat=7

< Report-To: {"endpoints":[{"url":"https:\/\/a.nel.cloudflare.com\/report\/v4?s=zAPZYPNT3vn%2F9jKV%2BBZor2BRWM7pwOcuv%2F4TB2TZHuH2UJGDK31EuHgbf3K69pbfszSU7T1gsPX5VWp5ltY7wX8nwfX1Wj6HPfQyRWOQ5%2FcwqX1RByyx9gjusPGNkxhn"}],"group":"cf-nel","max_age":604800}

< NEL: {"success_fraction":0,"report_to":"cf-nel","max_age":604800}

< Server: cloudflare

< CF-RAY: 8828947a8d573aea-IAD

< 

<html>

<head><title>301 Moved Permanently</title></head>

<body>

<center><h1>301 Moved Permanently</h1></center>

<hr><center>cloudflare</center>

</body>

</html>

* Connection #0 to host lexaloffle.com left intact
```

Aemiii91, 2024-05-12:
Visiting the url in a browser works fine 🤷‍♂️

https://www.lexaloffle.com/bbs/cpost_lister3.php?cat=7

Output:
[image]

LO-FANG, 2024-05-17:
Hello, I want to know if this is a problem with my machine or with the pico-8 app

XK9274, 2024-05-17:
> Hello, I want to know if this is a problem with my machine or with the pico-8 app

Have you removed any system files from your device, moved anything from it's correct/intended location?have you deleted any of the FSCK files from the root of the SD?

Can you open the terminal and check that `wget` runs ok?

LO-FANG, 2024-05-17:
> Have you removed any system files from your device, moved anything from it's correct/intended location? Have you deleted any of the FSCK files from the root of the SD?
>
> Can you open the terminal and [...]

I tried to fix the problem by reinstalling the system multiple times, but unfortunately, it didn't work. I didn't delete any system files, and wget is working fine.

LO-FANG, 2024-05-17:
> Have you removed any system files from your device, moved anything from it's correct/intended location? Have you deleted any of the FSCK files from the root of the SD?
>
> Can you open the terminal and [...]

Is your onion splore working fine? If it is, then the problem's on my end.

LO-FANG, 2024-05-20:
> > > > Hello, I want to know if this is a problem with my machine or with the pico-8 app
> > > 
> > > 
> > > Have you removed any system files from your device, moved anything from it's correct/intended location? Have you deleted any of the FSCK files from the root of the SD?
> > > Can you open the terminal and check that `wget` runs ok?
> > 
> > 
> > Is your onion splore working fine? If it is, then the problem's on my end.
> 
> It's working fine yes, if it wasn't working for everyone we'd have hundreds of issues about it.
> 
> Have you tried on a hotspot or another wireless network?

Hug, It worked for a while yesterday,today it not work again, I found out that when I open the Splore, the WIFI will disconnected.

LO-FANG, 2024-05-20:
> It's working fine yes, if it wasn't working for everyone we'd have hundreds of issues about it.
>
> Have you tried on a hotspot or another wireless network?

You may not believe this, it just fixed itself, wtf.

closed 2024-05-24 by XK9274