OnionUI/Onion #1611: Tweaks -> Network-features only save if you exit with the button B

Gammelstulle, 2024-06-19:
### Checklist

- [X] I have reviewed the [Onion Wiki](https://onionui.github.io/docs) and something is incorrect or missing

### Wiki Section

https://onionui.github.io/docs/network-features

### Description

Please add to the documentation of Samba, FTP, HTTP, (and all other network-features) that you **NEED** to press B to exit **and save** changes. 
If you otherwise long press the Menue button instead, your changes will not save.
TBH I was so confused why my settings I made for the "network-features" are not saved after set end exit via Menue button. (In the other hand, settings like "show expert mode" are saved just fine if you exit via the menue button.

Maybe this could be also a bug, otherwise my feature request for teh future would be that long press the menue button to exit also save the changes of all settings. (It is just a bit inconsistent the way how it is now)

### Correction

(add this to any network feature doc ore just add it to your FAQ)
### Save changes
Go back to the tweaks main menue.
Exit tweaks with B o apply & start the FTP server

Aemiii91, 2024-06-19:
I'll look into it. You shouldn't have to long press menu for it to exit (a single short press should be enough)

Gammelstulle, 2024-06-19:
> I'll look into it. You shouldn't have to long press menu for it to exit (a single short press should be enough)

Yikes. I new, use onion since 1 day x.x but love it a lot :).

Aemiii91, 2024-06-23:
Saves fine for me, when pressing MENU after changing a toggle in Tweaks -> Network.

closed 2026-01-20 by Aemiii91