OnionUI/Onion #1613: Menu+Vol shortcut not working for brightness in 4.3.1

cspez, 2024-06-23:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.3.1

### Miyoo Firmware Version

202306092122 and 202306282128

### Provide a clear and concise description of the issue

After upgrading from 4.3.0 to 4.3.1, the brightness shortcut of menu+vol is no longer working. The OSD bar on the left shows up and moves in the correct direction 1 unit plus/minus, but it immediately snaps back to the original location and the screen brightness does not change. I hear constant clicking (menu sound) while holding down the menu button when using the shortcut. Brightness can still be changed normally through the settings. Tried on two different MM+ devices (same SD card) and the same issue on both.

### Steps to Reproduce

Upgrade from 4.3.0 stable to 4.3.1 stable, use brightness shortcut.

### Provide any additional context or information that may help us investigate your issue

Have not tried a fresh install at this point.

Starlightbotanist, 2024-06-24:
I am able to reproduce this on my MMP.

Starlightbotanist, 2024-06-24:
Update, for me the issue only exists while on the OnionOS home screen. Brightness hotkeys work as expected in game. @cspez can you confirm this behavior on your end?

cspez, 2024-06-24:
I can confirm that in-game it works fine.

bn40400, 2024-06-24:
I can confirm this issue as well, in game works fine, but Main UI (Home) Hot key shortcut does not work.

PR #1616 (2024-06-24): Fix brightness and screenshot shortcuts in MainUI
Fix regression bug in `keymon` causing brightness and screenshot to not work in MainUI.

Fixes #1613
commit 7dfc008b (2024-06-24): Fix brightness and screenshot shortcuts in MainUI (#1616)

Fix regression bug in `keymon` causing brightness and screenshot to not
work in MainUI.

Fixes #1613
closed 2024-06-24 by Aemiii91