OnionUI/Onion #1625: Issue Title

DSkywalk, 2024-07-03:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

[AMSTRAD-CPC] Upgrade to lastest caprice32

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

Hello everyone!

I have seen in docs that you asked for help with the CPC and I am the official developer of cap32-libretro, I would like to help you in what you need because the emulator in recent years has experienced great improvements that greatly facilitates its use and enjoyment. 

The first thing I have seen is that the version you include is VERY old, currently the stable version is 4.5.4 and the latest versions also include a trimmed screen to enter perfectly in 640x480 screens.

[image]

In addition we have been working on a game database, called [clean-cpc-db](https://github.com/libretro/libretro-cap32/issues/112) (very easy to find a pack in archive) that allows you to click and play as it directly configures the keyboard to [play by pressing START](https://github.com/libretro/libretro-cap32/wiki/RMP:-Done) and not worry about anything else. As if it were a console.

[image]

I have put a lot of effort in bringing the CPC to the public and I would like you to have the best support because I love Onion :heart: 

### Steps to Reproduce

Current version suggest an abandoned __crocods__ that a basic emulation and needs the use of keyboard and a version with many years of __caprice32__.

### Provide any additional context or information that may help us investigate your issue

In short, my advice is to update the cap32_libretro build to the latest version which is perfectly stable. If you need any kind of help/support you have me watching this thread.

Amiga500, 2024-07-07:
@DSkywalk @schmurtzm Congratulations and thanks for your work on cap32-libretro and clean-cpc-db, i hope that it will add soon on Onion.

Amiga500, 2024-12-23:
Will anyone be offended if I give 50€ to whoever implements this in the new 4.4.0?

DSkywalk, 2024-12-23:
Well, I can do little more than offer myself for whatever they need, if you want @schmurtzm  I can compile the latest version and push it as a new PR :christmas_tree:

Aemiii91, 2024-12-23:
I have it compiled on my machine, currently testing ✅

Aemiii91, 2024-12-23:
@DSkywalk Seems this needs to be updated: https://github.com/libretro/libretro-super/blob/876011e1cfce82d931a0717a181be9b6401a9a80/dist/info/cap32_libretro.info#L8

Amiga500, 2024-12-23:
@Aemiii91 @DSkywalk Both of you will receive the prize

closed 2024-12-23 by Aemiii91

Amiga500, 2024-12-23:
@DSkywalk how can i reach you to pay?

DSkywalk, 2024-12-24:
> @DSkywalk how can i reach you to pay?

You could do any donation using my support :heart_decoration:  icon in my personal blog: http://david.dantoine.org/

Enjoy and Happy Christmas! :christmas_tree:

Amiga500, 2024-12-24:
Happy Christmas! 🎄

PR #1727 (2024-12-23): Update core: Caprice32 (includes screen crop option)
Closes #1625

Screen crop works as intended:
[image]
commit faea11bc (2024-12-23): Update core: Caprice32 (includes screen crop option) (#1727)

Closes #1625

Screen crop works as intended:
![Megablasters (1994)(Radical Software)(Disk 1 Side
A)_001](https://github.com/user-attachments/assets/6337e690-a624-476a-8aff-de69e0069eb6)