OnionUI/Onion #1631: Feature: MiyooMini+ should increment time on startup if Wifi is disabled on devices without RTC

BaseInfinity, 2024-07-16:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.3.0

### Miyoo Firmware Version

202306282128

### Provide a clear and concise description of the issue

Based off of issue here:
https://github.com/XK9274/syncthing-app-miyoo/issues/18

I've recently been using Syncthing on a MiyooMini+ to sync devices that does not have a RTC chip and this can cause file conflicts since Syncthing uses the date modified timestamp to decide if there's a conflict.

If you turn off your device, then turn it on without wifi, the clock will still read the same and the next time you save, you will create a conflict.

An easy solution could be to increment the system time by 1, but ONLY if wifi is disabled and a RTC is not on the device (I understand that newer MM+ don't have this issue due to RTC)

### Steps to Reproduce

Simply save a game and turn off the device.

Now with Wifi Disabled turn on the MM+ and save a new game, it will have the same timestamp as when you turned off the device, we froze time :O

### Provide any additional context or information that may help us investigate your issue

I know it's easy to just upgrade to latest MM+ but I thought this could be an easy solution and still allow those with older versions to use Syncthing

Aemiii91, 2024-07-16:
You can set the emulated time skip in **Tweaks -> System -> Date and time**. This is already a core Onion feature.

closed 2024-07-16 by Aemiii91

BaseInfinity, 2024-07-16:
@Aemiii91 Thanks for the quick response!

It seems that I cannot set the `Emulated Time Skip` as well as the `Set Automatically via internet` feature.

Will the Emulated Time skip still kick in if Wifi cannot be detected with `Set Automatically Via Internet` set?

Or can I only use one or the other?

BaseInfinity, 2024-07-16:
Also would it be possible to have a less dramatic `Emulated Time Skip`? +1 hour and more can potentially create `Saves` in the future. Maybe if it was a timeskip of seconds or minutes, it would help avoid the issue?

Aemiii91, 2024-07-16:
You have to choose which one to use. Skipping time and then connecting and synching time via wifi will probably cause issues with save states and activity tracking.

BaseInfinity, 2024-07-16:
> You have to choose which one to use. Skipping time and then connecting and synching time via wifi will probably cause issues with save states and activity tracking.

@Aemiii91 darn okay, I was hoping the Emulated Timeskip (of maybe +1 second) could be used as a "Fallback" feature when Wifi isn't detected on non RTC devices

You're right that this will probably create more headaches than it is worth to use Timeskip as a backup and using smaller intervals; Versus me just buying the latest MiyooMini+ lol

I can also just be more careful about my saves knowing this kind of limitation (like be sure to resave my game file with wifi on before switching devices)

BaseInfinity, 2024-07-16:
If I wanted to submit a change to use Emulated Time Skip as a fallback with smaller intervals, do you think that would be worthwhile for the project or do you think it's a waste of time (no pun intended =P) and not worth the headaches?

Aemiii91, 2024-07-16:
It's definitely way outside the intended use. But it shouldn't be too hard to add.