OnionUI/Onion #1699: Video player audio speed up when switching audio tracks

VictorMitr, 2024-11-07:
### Checklist

- [X] I am reporting a bug in the Onion OS - Any issues with the closed source frontend ([MainUI](https://github.com/OnionUI/MainUI-issues)), or [RetroArch](https://github.com/libretro/RetroArch) or any of the cores should be directed to their respective repositories
- [X] I have reviewed the [Onion Docs](https://onionui.github.io/docs) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.3.1-1

### Miyoo Firmware Version

202407211632

### Provide a clear and concise description of the issue

Hi, I have noticed a problem with the videos having more than 1 audio track and at least 1 subtitles track embedded into the file. When switching audio tracks for such files the video is speed up while the audio goes normal speed. I can surely say this issue is consistent for all video files with subtitles embedded.

I have confirmed that for similar video files with several audio tracks and without the subtitles, the issue disappears. I have tested it for similar video files having:

- Video resolution not more than 640x480p, AVC encoded (H.264), 23.98-30 fps
- Audio streams AAC LC codec, 2 channels, 48 kHz
- .mp4

### Steps to Reproduce

1. Have a video with a subtitles track and several audio streams in /Media/Videos folder
2. Open video player
3. Navigate to video, play video
4. Press "L" to change audio tracks

### Provide any additional context or information that may help us investigate your issue

Cannot provide the content, but one can convert any video with embedded subtitles and several audio track using Handbrake and this preset: 

[miyoomini_480p_subs_all-languages.json](https://github.com/user-attachments/files/17658184/miyoomini_480p_subs_all-languages.json)

Aemiii91, 2024-11-07:
The video player is ffplay and comes from: https://github.com/bobotrax/ffplay_Miyoo

We are not actively maintaining it, and the project seems stale at this point (no updates in 2 years).

closed 2024-11-07 by Aemiii91

Aemiii91, 2024-11-07:
Not at the moment